pybricks/support #814: Technic Hub can get stuck during shutdown

laurensvalk, 2022-11-26:
The Technic Hub can get stuck during power off.

This appears to happen when trying to shut down the hub when a Bluetooth task within a program is busy.

_Originally posted by @laurensvalk in https://github.com/pybricks/support/issues/600#issuecomment-1328101063_

laurensvalk, 2022-11-26:
https://user-images.githubusercontent.com/12326241/204106196-00c3a3a8-dc97-472c-aff9-d31b7af8cdcc.mp4

laurensvalk, 2022-11-26:
I have a hunch we are hitting something like this: https://github.com/pybricks/support/issues/567, which prevents the program from ending, and thus preventing poweroff.

laurensvalk, 2022-11-26:
See also the **problem** section in https://github.com/pybricks/support/issues/802

laurensvalk, 2022-11-26:
I was able to reproduce this several times (but not always) under the following conditions:

- Laptop scanning for hubs in Pybricks Beta
- Workstation scanning for hubs in Pybricks Beta
- They were about 3 meters apart
- A Prime Hub, an Inventor Hub, a Move Hub, and a Technic Hub all advertising
- Connect to the Technic Hub and run [this](https://docs.pybricks.com/en/latest/pupdevices/remote.html#checking-which-buttons-are-pressed) program.

Sometimes it works, sometimes it does not. You can tell that it happens when the remote is off but the hub keeps running without hitting the timeout.

Turning off then will trigger this.

pbochynski, 2022-11-26:
I just had the same issue 10 minutes ago. Had to cut the power off to get rid of the blue flashlight.

laurensvalk, 2022-11-27:
Was this in a program doing something with Bluetooth, like the remote or broadcasting/scanning?

pbochynski, 2022-11-28:
It happened only once to me. I do not remember exactly if it was during scanning or remote was already connected. Next time I will note it down, but I did a lot of experiments with remote trying to fix https://github.com/pybricks/support/issues/813, and I couldn't reproduce it.

closed 2022-12-08 by laurensvalk