pybricks/support #824: [Bug] No traceback after press of stop-button

BertLindeman, 2022-12-03:
**Describe the bug**
In beta.pybricks.com running firmware `3.2.0b6` traceback is no longer shown after pressing the stop-button.
So you no longer see what source line the stop was effected. 

**To reproduce**
Steps to reproduce the behavior:
1. Go to 'beta.pybricks.com'
2. Pair with a Robot inventor or Technic hub with firmware 3.2.0b6
3. Load and run program (see below) 
4. Press the stop button
5. See the output:

On firmware 3.2.0b6:

```
Running ('technichub', '3.2.0b6', 'v3.2.0b6 on 2022-12-02')
Stop button pressed.
```

**Expected behavior**
See the traceback as it was before 3.2.0b6 like on firmware 3.2.0b5:

```
Running ('primehub', '3.2.0b5', 'v3.2.0b5 on 2022-11-11')
Traceback (most recent call last):
  File "simple_wait.py", line 12, in <module>
SystemExit: stop button pressed
```

**Screenshots**
Not now.

**Program source**
```python

# from pybricks.hubs import TechnicHub
# from pybricks.pupdevices import Motor
# from pybricks.parameters import Button, Color, Direction, Port, Stop
# from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch
from pybricks import version

# hub = TechnicHub()

print("Running", version)

wait(50000)

```

laurensvalk, 2022-12-03:
We have hidden the traceback when pressing the stop button. In almost all cases, the traceback on stop just adds clutter to the output window. Additionally, new users sometimes wonder why they see errors when there is nothing wrong.

All other exceptions are still printed with the full traceback to help you find errors in your script.

Did you have a particular use case for seeing the traceback on pressing stop?

If you really want the traceback for stopping the program, you can still press CTRL+C in the I/O window.

This seemed like the right balance, but let us know if we missed something :)

BertLindeman, 2022-12-03:
I was just running a bit larger program and as I was used, pressed the stop-button and expected to see the current source line.
In that program having trouble with 'stalled()' and 'run_until_stalled()' but that could also be caused by the gearbox in the model . 

> If you really want the traceback for stopping the program, you can still press CTRL+C in the I/O window.

That seems a good balance indeed.

This question can be closed if you like.

laurensvalk, 2022-12-03:
As always, your feedback is highly appreciated. Thanks for testing the new beta so quickly!

laurensvalk, 2022-12-03:
We did change quite a few things to stall behavior behind the scenes. But it is supposed to work about the same as before.

Let us know if you see any discrepancies!

closed 2022-12-03 by laurensvalk

BertLindeman, 2022-12-21:
Seems that I can not get a CTRL_C through on the IO window running beta.pybricks v3.2.0 (Pybricks Code v2.0.0).

Click in the IO window shows a blinking cursor, so it looks like the window *is* selected.
But pressing CTRL-C does nothing.
The program above is running the
```python
wait(50000)
```
on the CTRL-C press.

Should I start a new issue?

laurensvalk, 2022-12-21:
Are you able to type something in the REPL? Is this similar to #853?

BertLindeman, 2022-12-21:
Can start REPL but no keyboard input seen or accepted.
Not even CTRL-C nor CTRL-D

Need to press "Stop program" button to stop REPL.
Huh, but pressing F6 **does** work to stop REPL.

Not sure that I understand #853

laurensvalk, 2022-12-21:
Thank you for testing! I think the unable-to-enter-text part is the same as #853.

Now we know that we should investigate it with Edge.

The disconnecting in #853 may be a separate issue, but let us know if you see the same issue with the technic hub.

BertLindeman, 2022-12-21:
Thank you, Laurens.
Will take a better look at #853