pybricks/support #829: [Feature] Restore frozen directory to include frozen modules

laurensvalk, 2022-12-06:
We used to have a folder called "frozen" in the each of the bricks/ folders, and something in the Makefile that would freeze all of the contents into the firmware. We can probably restore that. It shouldn't take any space for anyone who isn't using that feature.

_Originally posted by @laurensvalk in https://github.com/pybricks/support/discussions/828#discussioncomment-4322426_

laurensvalk, 2022-12-06:
This could be useful for large modules such as those made by @novakasa inhttps://github.com/orgs/pybricks/discussions/828#discussioncomment-4322426.

laurensvalk, 2022-12-06:
> It shouldn't take any space for anyone who isn't using that feature.

Not quite true since it requires enabling `MICROPY_MODULE_FROZEN`

laurensvalk, 2023-01-13:
This is a first step:

```diff
diff --git a/bricks/_common_stm32/make.mk b/bricks/_common_stm32/make.mk
index b9d8e32e..9c12a968 100644
--- a/bricks/_common_stm32/make.mk
+++ b/bricks/_common_stm32/make.mk
@@ -50,6 +50,9 @@ USER_C_MODULES = $(PBTOP)
 
 include ../../micropython/py/mkenv.mk
 
+# Frozen Python code
+FROZEN_MANIFEST ?= ../_common_stm32/manifest.py
+
 # qstr definitions (must come before including py.mk)
 QSTR_DEFS = ../_common/qstrdefs.h
 QSTR_GLOBAL_DEPENDENCIES = ../_common/mpconfigport.h ../_common_stm32/mpconfigport.h
@@ -353,6 +356,17 @@ SRC_QSTR += $(PY_EXTRA_SRC_C) $(PY_STM32_SRC_C) $(PYBRICKS_PYBRICKS_SRC_C)
 # Append any auto-generated sources that are needed by sources listed in SRC_QSTR
 SRC_QSTR_AUTO_DEPS +=
 
+ifneq ($(FROZEN_MANIFEST),)
+CFLAGS += -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool
+CFLAGS += -DMICROPY_MODULE_FROZEN_MPY
+CFLAGS += -DMPZ_DIG_SIZE=16
+MPY_TOOL_FLAGS += -mlongint-impl none
+endif
+
+ifneq ($(FROZEN_MANIFEST),)
+CFLAGS += -DMICROPY_MODULE_FROZEN_STR
+endif
+
 # Main firmware build targets
 TARGETS := $(BUILD)/firmware.zip
```

But we'd also need some work to our import function to facilitate this.

Novakasa, 2023-01-13:
I actually tried something very similar to this yesterday:
https://github.com/Novakasa/pybricks-micropython/tree/restore-frozen

When trying to compile a frozen module, I get the error:
[image]

And when no modules are present (e.g. no lines in manifest.py or no .py in frozen `modules` folder:
[image]

Is that what you mean be some work left on the import function or am I seeing a different issue?

laurensvalk, 2023-01-13:
There's a bit more to it. Trying now, hopefully will report back soon :)

laurensvalk, 2023-01-13:
I didn't finish the import part yet, but the following might be good enough for you. It will just always run `bricks/_common_stm32/modules/my_code.py` just before your script starts.

Add these files:

`bricks/_common_stm32/manifest.py`
```
freeze_as_mpy("modules")
```

`bricks/_common_stm32/modules/my_code.py`
```python
print("hello, data!")
data = 123
```

And make these changes:

```diff
diff --git a/bricks/_common_stm32/make.mk b/bricks/_common_stm32/make.mk
index b9d8e32e..9c12a968 100644
--- a/bricks/_common_stm32/make.mk
+++ b/bricks/_common_stm32/make.mk
@@ -50,6 +50,9 @@ USER_C_MODULES = $(PBTOP)
 
 include ../../micropython/py/mkenv.mk
 
+# Frozen Python code
+FROZEN_MANIFEST ?= ../_common_stm32/manifest.py
+
 # qstr definitions (must come before including py.mk)
 QSTR_DEFS = ../_common/qstrdefs.h
 QSTR_GLOBAL_DEPENDENCIES = ../_common/mpconfigport.h ../_common_stm32/mpconfigport.h
@@ -353,6 +356,17 @@ SRC_QSTR += $(PY_EXTRA_SRC_C) $(PY_STM32_SRC_C) $(PYBRICKS_PYBRICKS_SRC_C)
 # Append any auto-generated sources that are needed by sources listed in SRC_QSTR
 SRC_QSTR_AUTO_DEPS +=
 
+ifneq ($(FROZEN_MANIFEST),)
+CFLAGS += -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool
+CFLAGS += -DMICROPY_MODULE_FROZEN_MPY
+CFLAGS += -DMPZ_DIG_SIZE=16
+MPY_TOOL_FLAGS += -mlongint-impl none
+endif
+
+ifneq ($(FROZEN_MANIFEST),)
+CFLAGS += -DMICROPY_MODULE_FROZEN_STR
+endif
+
 # Main firmware build targets
 TARGETS := $(BUILD)/firmware.zip
 
diff --git a/pybricks/pybricks.c b/pybricks/pybricks.c
index 8bf141aa..50bf0980 100644
--- a/pybricks/pybricks.c
+++ b/pybricks/pybricks.c
@@ -2,6 +2,7 @@
 // Copyright (c) 2018-2022 The Pybricks Authors
 
 #include "py/mpconfig.h"
+#include "shared/runtime/pyexec.h"
 #include "py/obj.h"
 #include "py/objmodule.h"
 #include "py/objstr.h"
@@ -122,6 +123,7 @@ void pb_package_pybricks_init(bool import_all) {
         if (import_all) {
             pb_package_import_all();
         }
+        pyexec_file_if_exists("my_code.py");
         nlr_pop();
     } else {
         // Print initialization or import exception.
```

Novakasa, 2023-01-13:
Thanks for this! Unfortunately this still seems to have the same problem as I had yesterday: `error compiling <my file>` `OSError: 2` I have done a clean compilation, but maybe something in my environment is outdated or something.

laurensvalk, 2023-01-13:
Here is a ready made version that you can try: https://github.com/pybricks/pybricks-micropython/commits/frozencode

Importing also works now. You should be able to put any number of files in the modules folder. There is one example to start with.

I'll clean it up to make it properly optional later.

laurensvalk, 2023-01-13:
I've tested this on Prime Hub only, but it should also work on City Hub. It builds fine for both at least.

Novakasa, 2023-01-13:
Yeah unfortunately I still get this `OSError: 2`. I'll see if I can setup my environment again or test on ubuntu these next few days. Thanks for working on this!!

laurensvalk, 2023-01-13:
Your CI seems to be producing artifacts, so you could push some modules and download the firmware from there.

laurensvalk, 2023-01-14:
This is done now. See usage instructions in https://github.com/pybricks/pybricks-micropython/tree/master/bricks/_common/modules.

If you want to use CI approach @Novakasa, make sure to add exceptions for your files to [.gitignore](https://github.com/pybricks/pybricks-micropython/tree/master/bricks/_common/modules/.gitignore) to e.g. change it to:
```
*.py
!test.py
```

laurensvalk, 2023-01-14:
> It shouldn't take any space for anyone who isn't using that feature.

This is achieved by enabling the extra import code only if frozen modules are present.

closed 2023-01-14 by laurensvalk

Novakasa, 2023-01-14:
[image]
works using the CI builds 👍

laurensvalk, 2023-01-14:
Yes! Now, how many lines of code will you insert, and how much download-and-run time does this save you? :smile:

Novakasa, 2023-01-14:
I'm inserting 190 lines for my communication module, and my train program download goes from ~5kB to ~2kB. I'll keep experimenting on how much more complexity I am not able to handle in my train program, which could make some exciting things possible!