pybricks/support #833: beta.pybricks.com to code.pybricks.com file migration

dlech, 2022-12-07:
When we release the next version of Pybricks Code to `code.pybricks.com`, users who have been using `beta.pybricks.com` will likely want to move their files. Exporting the files is easy using the *Backup all files* button. But importing doesn't currently accept zip files, so the backup needs to be unzipped first. The *Import a file* open dialog allows multiple selection, so all files from this folder can be imported at one time.

So we should either document how to do this or consider allowing selecting a zip file when importing.

laurensvalk, 2022-12-07:
> allowing selecting a zip file when importing.

I like this idea, but I'd be OK to release without this feature.

Copying a few files over isn't that much work for the end user, and not a bad idea anyway when migrating to a stable version. The beta isn't going away, so there is no risk of losing files.

dlech, 2023-02-13:
This issue hasn't really come up, so we probably don't need to spend time on this.

laurensvalk, 2023-02-14:
> allowing selecting a zip file when importing.

This could be useful if people use the download feature to move their stuff to another computer.

And edge cases like https://github.com/pybricks/support/issues/896#issuecomment-1429264523

laurensvalk, 2023-02-24:
I'm re-opening this based on input from an FLL coach. Another good use case for this is allowing FLL/WRO teams to back up their work and share between users/computers:

_Originally posted by  @TheWendyPower in https://fosstodon.org/@WendyDLN@mastodon.online/109909743109786344_:

> I have a quick question for you, and if it is in the documentation please direct me there. Can the programs be stored locally without have to download them. Right now we sync our code between two laptops and a backup with SyncThing. Not loosing all the code is the idea.

_Originally posted by  @laurensvalk  in https://fosstodon.org/@laurensvalk/109912897687287118_:

> Download is the only way at present.
> (...)
> On a smaller scale, could we make it better by letting you upload the same .zip again?

_Originally posted by  @TheWendyPower in https://fosstodon.org/@WendyDLN@mastodon.online/109915280010666159_:

> Yes, being able to upload the same zip files would be great. We would have the kids download the file at the end of each meeting, and into a folder that syncs. This would allow them to have a backup of their code still, switch laptops, and collaborate from different locations. Backup is the most important.

closed 2023-03-07 by dlech