pybricks/support #834: regression: stop button does not run finally statement

dlech, 2022-12-07:
**Describe the bug**
Since we started using `SystemAbort` instead of `SystemExit` when the stop button is pressed, the `finally` clause of `try` statements is no longer run.

**To reproduce**

Run this program and press the stop button after "ready" is printed.

```python
from pybricks.pupdevices import ColorSensor
from pybricks.parameters import Port
from pybricks.tools import wait

sensor = ColorSensor(Port.A)

try:
    sensor.lights.on(20)
    print("ready")

    while True:
        wait(1000)

finally:
    sensor.lights.off()
```

**Expected behavior**
Programs should be able to "clean up" after the stop button is pressed as they did in previous releases.

The stop button should only be `SystemExit` while power off should be `SystemAbort`.

If users make a program that manages to always catch `SystemExit`, they can still power off the hub to get out of the program.

laurensvalk, 2022-12-07:
We can just revert https://github.com/pybricks/pybricks-micropython/commit/016f3a9c81afd2dee7f8ef974c273768b7f73c5c

laurensvalk, 2022-12-07:
Nitpick for revert commits, can we take a few seconds to format the commit title properly? :)

dlech, 2022-12-07:
What is "properly" for reverts? The autogenerated subject seems fine to me.

laurensvalk, 2022-12-07:
Starting with the path just like every other commit would be nice.

Could drop the quotes too, but I suppose that's just preference.