pybricks/support #838: Compatibility icons don't scale well for narrow-width docs

dlech, 2022-12-07:
As seen in the screenshot, when the docs are viewed on a narrow-width frame, the compatibility icons become too small to see clearly.

[image]

To make it more clear, the checkmark or exmark should be nearly as large as the entire image.

The images completely disappear when the docs are too narrow.
[image]

laurensvalk, 2022-12-08:
Now I just need a way to get the css right to get some padding on the background image.

laurensvalk, 2022-12-09:
End result:

[image]

small screen:

[image]

closed 2022-12-09 by laurensvalk