pybricks/support #851: [Bug] Pybricks Code connection gets stuck halfway

laurensvalk, 2022-12-12:
**Describe the bug**
Quite frequently my hub (Prime/Technic) gets through the initial connection phase only partially.

The hub light turns solid blue (indicating a connection), but the Play/REPL buttons remain disabled.

Then I try to connect again with the Bluetooth Button. The hub shows up there as "paired". This isn't supposed to work, but connecting to that usually does make it work (?).

**To reproduce**
I don't know yet.

I'm using Ubuntu 22.04 with Google Chrome.

laurensvalk, 2022-12-12:
This was also reported by @mattytrentini.

laurensvalk, 2022-12-12:
This also happens on Windows.

laurensvalk, 2022-12-12:
it can also get stuck in this state, possibly different issue.

[image]

dlech, 2022-12-12:
Anything in the Chome console? Wireshark logs?

dlech, 2023-03-24:
does this still happen?

laurensvalk, 2023-03-25:
I think so, occasionally.

laurensvalk, 2024-10-02:
I haven't seen this happen in a long time, so let's close this.

closed 2024-10-02 by laurensvalk