pybricks/support #854: [Bug] Prime Hub does not stop advertising on connecting

laurensvalk, 2022-12-12:
**Describe the bug**
The Prime Hub does not stop advertising when connecting/running a program.

This seems to cause some of the behaviors we are seeing in #849, etc.

Some of these problems go away if we explicitly stop advertising:

```diff
diff --git a/lib/pbio/sys/bluetooth.c b/lib/pbio/sys/bluetooth.c
index 0d699db1..cb62e989 100644
--- a/lib/pbio/sys/bluetooth.c
+++ b/lib/pbio/sys/bluetooth.c
@@ -266,10 +266,10 @@ PROCESS_THREAD(pbsys_bluetooth_process, ev, data) {
             || pbsys_status_test(PBIO_PYBRICKS_STATUS_USER_PROGRAM_RUNNING)
             || pbsys_status_test(PBIO_PYBRICKS_STATUS_SHUTDOWN));
 
-        if (!pbdrv_bluetooth_is_connected(PBDRV_BLUETOOTH_CONNECTION_LE)) {
+        // if (!pbdrv_bluetooth_is_connected(PBDRV_BLUETOOTH_CONNECTION_LE)) {
             // if a connection wasn't made, we need to manually stop advertising
             pbdrv_bluetooth_stop_advertising();
-        }
+        // }
 
         pbsys_status_clear(PBIO_PYBRICKS_STATUS_BLE_ADVERTISING);
```

dlech, 2022-12-12:
We should be able to fix this in the BTStack Bluetooth driver.

dlech, 2022-12-12:
I'm not able to reproduce the problem. How do you know the hub is still advertising?

laurensvalk, 2022-12-12:
To reproduce, start a program from Pybricks Code and then disconnect. When you click the Bluetooth icon again, the hub still shows in the dialog (always, so I'm not pushing for special timing here).

With the aforementioned patch/hack on the firmware side, it doesn't show up after disconnecting when the program is still running - then it only comes back when the program ends.

I'll try going back to see when this started happening for me.

dlech, 2022-12-12:
> To reproduce, start a program from Pybricks Code and then disconnect.

That is issue #849.

This issue says:

> The Prime Hub does not stop advertising when connecting/running a program.

which is different.

dlech, 2022-12-12:
OK, let's just close this as a duplicate then.

closed 2022-12-12 by dlech