pybricks/support #856: [Bug] Clicking Pybricks Code menu pane does not give it focus.

laurensvalk, 2022-12-13:
**Describe the bug**
Clicking anywhere in the Pybricks menu pane does not give it focus.

As a result, the various file shortcuts or search do not work.

It works after you click a button or file name inside the pane.

**To reproduce**

https://user-images.githubusercontent.com/12326241/207285836-314138d2-568c-4766-9077-377326e96ad1.mp4

**Expected behavior**
Focus (e.g. get blue outline) after clicking in menu pane.

dlech, 2022-12-13:
I had done something like this before, but removed it in https://github.com/pybricks/pybricks-code/commit/c3fec0b77996c775a6838ca602155fc6fec5ea03. Apparently it broke dialogs.

dlech, 2022-12-16:
I decided to just drop this styling altogether instead.

laurensvalk, 2022-12-16:
Aside from the styling, this issue was more about that it is somewhat hard to get focus without opening another file, just in order to use the search function.

I didn't know we had a search function until I went through the translation strings :smile: 

Totally OK with not having this for V3.2 though :+1:

dlech, 2022-12-16:
Got it. If we narrow the scope to clicking the empty area of the file list activates focus on the last focused file, it should be fairly easy to implement.

closed 2022-12-16 by dlech