pybricks/support #860: [Bug] Incomplete error message for syntax error via F2

laurensvalk, 2022-12-15:
**Describe the bug**

When running a bad program, we now get a nicer message with the option to go to that line (top)

But if checking a bad program with `F2`, we do not (bottom).

[image]

**Expected behavior**
The same error in both cases (the one in the top of the screenshot).

laurensvalk, 2022-12-16:
I think this also just checks the main file.

laurensvalk, 2022-12-16:
It looks like we can fix this by making F2 compile in the same way as we do for run, but I'm not sure how we can do that.

dlech, 2022-12-16:
Also related: #695 (discusses if we should even have F2 as a feature at all).