pybricks/support #862: [Feature] Link to DFU instructions on hub not found

laurensvalk, 2022-12-20:
**Is your feature request related to a problem? Please describe.**
On failed DFU connection, we show this:

[image]

**Describe the solution you'd like**
Link to the instructions, just like we do in case of Bluetooth not found.

laurensvalk, 2022-12-20:
I gave this a shot. Works, but not quite right. It leaves the notification open.

I tried modeling it after `NoHub.tsx` first, but 

`onAction('flashFirmware')` 

looked different than

 `dispatch(firmwareDfuWindowsDriverInstallDialogDialogShow())`.

Is this a different approach, or maybe I missed the equivalent form for this one?

[image]

dlech, 2022-12-20:
Using `onAction()` defers the responsibility of calling `dispatch` to the caller of `put(alertsShowAlert())`. You then have to wait for `alertsDidShowAlert` to get the result of the users action and act accordingly.

closed 2022-12-20 by dlech