pybricks/support #865: [bug] Can start REPL but no keyboard input seen or accepted.

laurensvalk, 2022-12-21:
Can start REPL but no keyboard input seen or accepted.
Not even CTRL-C nor CTRL-D

Need to press "Stop program" button to stop REPL.
Huh, but pressing F6 **does** work to stop REPL.

_Originally posted by @BertLindeman in https://github.com/pybricks/support/issues/824#issuecomment-1361103826_

laurensvalk, 2022-12-21:
If you have the time to capture Bluetooth traffic as you try to enter text in the REPL, that might help a lot.

https://github.com/orgs/pybricks/discussions/270

Also, do you see any errors in the developer tools console log of the browser?

laurensvalk, 2022-12-21:
The part about F6 working is expected. It's the same as pressing the stop button. This command is not sent over the REPL, so it still gets through.

laurensvalk, 2022-12-21:
Does the REPL still work via code.pybricks.com?

BertLindeman, 2022-12-21:
Yes:
```
MicroPython v3.1.0 on 2021-12-16; Powered Up City Hub with STM32F030RC
Type "help()" for more information.
>>> whatthe heck
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 
```

BertLindeman, 2022-12-21:
All running win11 with Microsoft Edge
```Version 108.0.1462.54 (Official build) (64-bit)```

laurensvalk, 2022-12-21:
Does the hub type make a difference or is the REPL broken on Edge in Pybricks 3.2 for all hubs?

To save you time, spike and mindstorms are all considered the same.

Technic and city are also considered the same.

laurensvalk, 2022-12-21:
Great news that V3.1 works. That means we can probably fix it :smile:

laurensvalk, 2022-12-21:
Maybe Bluetooth is fine but the terminal pane isn't getting focus... I'll give you some more troubleshooting ideas later this afternoon. Thanks for the input so far!

BertLindeman, 2022-12-21:
Then I will drop mu attempt to trace BT for now.

The IO screen does get the focus with the mouse  selecting.

Not related, but the IO window does not scroll very nice. (Will keep that in mind for later too)

BertLindeman, 2022-12-21:
More good news:
```
Running ('primehub', '3.2.0', 'v3.2.0 on 2022-12-20')
Traceback (most recent call last):
  File "issue_824.py", line 19, in <module>
KeyboardInterrupt: 
Pybricks MicroPython v3.2.0 on 2022-12-20; SPIKE Prime Hub with STM32F413VG
Type "help()" for more information.
>>> 
```

BertLindeman, 2022-12-21:
Rats....

Laurens, I closed and opened a new EDGE window.
Now I can not recreate this issue.
No errors seen on developers-tools both before and now.

Suggest closing.

Sorry for your time.

laurensvalk, 2022-12-21:
Can you paste text in it via right click?

BertLindeman, 2022-12-21:
Funny.....
Connected Technic hub
copied some text "wait(50000)"
clicked IO window
right-click and select "Paste" a few times
Nothing happens....
Then click REPL button and see:
```
Pybricks MicroPython v3.2.0 on 2022-12-20; LEGO Technic Hub with STM32L431RC
Type "help()" for more information.
>>> wait(50000)wait(50000)wait(50000)
```

BertLindeman, 2022-12-21:
Ooops,
Now no reaction to CTRL-C again. ;-(

laurensvalk, 2022-12-21:
> Funny.....
> Connected Technic hub
> copied some text "wait(50000)"
> clicked IO window
> right-click and select "Paste" a few times
> Nothing happens....
> Then click REPL button and see:
> ```
> Pybricks [...]

This is currently known behavior, separate from this issue.

BertLindeman, 2022-12-21:
Ran out of inspiration.
Will skip this for a few days, due to lack of free time.

BertLindeman, 2022-12-21:
Still another try....

Re-using a beta.pybricks.com tab after "some time"?  hour?
There was no hub connected during idle status (also not on other tabs or browsers)

1.  connect hub: OK
2.  start issue_824.py
3.  click IO window to give it the focus (cursor blinks)
4.  CTRL-C or CTRL-D not accepted
5.  F6 to stop the program OK
6.  start REPL
7.  paste some text in the IO window
8.  text not seen in IO window

Now press "Refresh" to restart the MSEDGE tab
1.  power-off and -on the hub
2.  connect hub: OK
3.  start issue_824.py
4.  click IO window to give it the focus (cursor blinks)
5.  CTRL-C or CTRL-D : OK

Works again.

Maybe fiction, but could it something link this (just phantasizing):
An MSEDGE quirk? Puts a tab in idle mode and on revival does miss to set some bit or variable or . . . ?

dlech, 2022-12-21:
It looks like there are 2 issues here. 

1. Scrolling doesn't work anymore in the terminal. This is reproducible 100% of the time.
2. Sometimes keyboard input doesn't work at all in the terminal. This is difficult to reproduce. (I haven't found a reliable way to reproduce it yet, but I've seen it once so far  in Chromium on Linux running the dev version of Pybricks Code locally.)

dlech, 2022-12-21:
I was able to reproduce this once more. Not still not sure how I triggered it, but when it happened, I did some profiling with the Chrome developer tools and could see that this line ran:

https://github.com/pybricks/pybricks-code/blob/2f4811e7426f7f2d401072fc6105920aa502a4c5/src/terminal/Terminal.tsx#L173

But I didn't see the corresponding handler run:

https://github.com/pybricks/pybricks-code/blob/2f4811e7426f7f2d401072fc6105920aa502a4c5/src/terminal/sagas.ts#L51

So perhaps the while loop in the second link is getting stuck somehow.

dlech, 2022-12-21:
Fix released in Pybricks Code v2.0.1.

closed 2022-12-21 by dlech

BertLindeman, 2022-12-21:
Great, seen working on win11 in CHrome and EDGE.

Thank you!