pybricks/support #867: [Bug] terminal slider slides app up and down on Android

laurensvalk, 2022-12-21:
**Describe the bug**
https://user-images.githubusercontent.com/12326241/208987302-b2202c3f-5af5-4fbd-ae26-6867e228395c.mp4

It takes the whole app up and down. I don't think it did this before. 

It also isn't very smooth.

The horizontal doc slider works fine and smooth.

dlech, 2022-12-22:
I think this is an upsteam Chromium bug. It seems to only be a problem when used in full screen. Visiting the site in a regular browser works fine. 

We have the hack from https://css-tricks.com/the-trick-to-viewport-units-on-mobile/ in place. I removed it and it seems like it is no longer necessary. So I'm guessing it was fixed in Chromium but they botched the full screen side of things.

dlech, 2022-12-22:
I think we can work around this though by using "standalone" instead of "fullscreen". [This](https://web.dev/learn/pwa/app-design/#fullscreen-experience) says that "fullscreen" should be for "immersive experiences" like games, etc and we should probably be using "standalone"

closed 2022-12-22 by dlech