pybricks/support #869: [Feature] State of iOS support

laurensvalk, 2022-12-21:
We don’t officially support iOS at the moment, but I’m opening an issue just to see where we are. There are a few browser apps with web BLE support.

I think the WebBLE browser worked once upon a time, but not anymore. This app is open source but the version in the App Store is a paid version.

[image]

Then there is Bluefy, which is non-paid, but probably non-free.

[image]

dlech, 2022-12-21:
Can you use *Copy error message* button and paste it here so we can see the full stack trace?

laurensvalk, 2022-12-21:
That’s all there is I’m afraid.

```
@https://beta.pybricks.com/static/js/main.e2874b89.js:2:401046
f@https://beta.pybricks.com/static/js/main.e2874b89.js:2:1562982
@https://beta.pybricks.com/static/js/main.e2874b89.js:2:1562732
h@https://beta.pybricks.com/static/js/main.e2874b89.js:2:1470437
u@https://beta.pybricks.com/static/js/main.e2874b89.js:2:1470926
@https://beta.pybricks.com/static/js/main.e2874b89.js:2:1465158
promiseReactionJob@[native code]
```

I can try digging up some more later in the week.

dlech, 2022-12-21:
> I think the WebBLE browser worked once upon a time, but not anymore.

Needs at least:
- https://github.com/daphtdazz/WebBLE/pull/52
- https://github.com/daphtdazz/WebBLE/pull/55 (proper implementation would be better than polyfill)

laurensvalk, 2022-12-22:
See also https://github.com/daphtdazz/WebBLE/issues/42 :smile:

laurensvalk, 2022-12-22:
See also https://github.com/bricklife/Scrub/issues/14