pybricks/support #870: "The program was stopped" message should start with a newline

dlech, 2022-12-21:
**Describe the bug**
When a SystemExit or SystemAbort is raise, we print a special message that hides the stack trace. If there is already text on a line in a terminal, this message is just appended to that line which makes it hard to read.

**To reproduce**

Firmware v3.2.0

Steps to reproduce the behavior:
1. Connect to a hub in Pybricks Code
2. Start the REPL
3. Type something but don't press enter
4. Press the stop button

**Expected behavior**
The message should be on a new line.

**Screenshots**
[image]

laurensvalk, 2022-12-22:
Since this mainly happens during REPL usage, perhaps we could do something like:

```diff
diff --git a/bricks/_common/micropython.c b/bricks/_common/micropython.c
index 4c97f428..7cc98ddc 100644
--- a/bricks/_common/micropython.c
+++ b/bricks/_common/micropython.c
@@ -116,6 +116,7 @@ static void run_repl() {
         // Print which exception triggered this.
         print_final_exception(nlr.ret_val);
     }
+    mp_print_str(&mp_plat_print, "\n");
 }
 #endif
```

dlech, 2022-12-22:
I would rather make is universally the same. I often write programs with, e.g. `print(..., sep=' ')`.

laurensvalk, 2024-10-02:
Adding a simple `\n` would fix this but it makes the overal output of multiple runs a bit harder to read, since the stop message is usually one line removed from the corresponding program output.

[image]

I'm not sure if a fancier solution like a conditional break is worth it though, so maybe the above is OK?

BertLindeman, 2024-10-02:
A traceback message would also be printed at the location the cursor happens to be.
I will not object against a "\n" but will also not complain about no newline.

dlech, 2024-10-02:
We could use ANSI escape codes to change the color of the SystemExit message to help break things up visually.

laurensvalk, 2025-11-02:
Just in time for the third anniversary of this issue 😎 

This is fixed by having flush write `\r\n` if the last write did not include it. This also fixes interruptions of long prints:

[image]

closed 2025-11-02 by laurensvalk