pybricks/support #871: Motor(Port.D) freezes hub. (technic hub running hub to hub experimental firmware on beta.pybricks.com)

AngusCrump, 2022-12-22:
**Describe the bug**
What is the problem?
Initialising port D as motor causes hub to freeze.

**To reproduce**
Steps to reproduce the behavior:
Run any code with Motor(Port.D) in it, or run Motor(Port.D) in REPL.

**Expected behavior**
What did you expect to happen instead?
Port D is Initialised as a motor.

**Screenshots**
[image]

dlech, 2022-12-22:
I see you are using a custom firmware v3.2.0b4-58-g84e5308d. I'm guessing this is the experimental Bluetooth hub-to-hub firmware. It has port D disabled for debugging.

AngusCrump, 2022-12-22:
Ah, I see. I'm using the bluetooth hub to hub but I could probably find a workaround. The hub to hub communication works very well, do you have any idea when it will come out in an offical release?

closed 2022-12-22 by AngusCrump

dlech, 2022-12-22:
We don't have an exact timeline right now.