pybricks/support #872: [Bug] Unhandled error event in volume.js

laurensvalk, 2022-12-22:
I noticed the following in a local build of Pybricks Code.

Could this be related to https://github.com/pybricks/pybricks-code/commit/64b6251c1a220db9ed721a4cff37c0cd001210fe?

```
Compiled successfully!

You can now view @pybricks/pybricks-code in the browser.

  Local:            http://localhost:3000
  On Your Network:  http://192.168.1.108:3000

Note that the development build is not optimized.
To create a production build, use yarn build.

webpack compiled successfully
No issues found.
node:events:491
      throw er; // Unhandled 'error' event
      ^

TypeError: Cannot set property closed of #<Readable> which has only a getter
    at FsReadStream.close (/home/laurens/git-pybricks/pybricks-code/node_modules/memfs/lib/volume.js:2047:17)
    at FsReadStream._destroy (/home/laurens/git-pybricks/pybricks-code/node_modules/memfs/lib/volume.js:2032:10)
    at _destroy (node:internal/streams/destroy:109:10)
    at class_1.destroy (node:internal/streams/destroy:71:5)
    at class_1.<anonymous> (/home/laurens/git-pybricks/pybricks-code/node_modules/memfs/lib/volume.js:1962:22)
    at class_1.emit (node:events:525:35)
    at endReadableNT (node:internal/streams/readable:1359:12)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
Emitted 'error' event on class_1 instance at:
    at emitErrorNT (node:internal/streams/destroy:151:8)
    at emitErrorCloseNT (node:internal/streams/destroy:116:3)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)

Node.js v18.12.1
laurens@laurens-laptop:~/git-pybricks/pybricks-code$
```

laurensvalk, 2022-12-22:
To reproduce:
- Open the firmware install menu
- Pick technic hub
- Proceed to the fourth (last) step, no need to do anything here
- Pybricks Code has crashed

dlech, 2022-12-22:
The crash you shared is in the tooling, not the actual app. 

> Node.js v18.12.1

We are currently using node v16. So the crash is likely due to the different node version. See https://github.com/pybricks/pybricks-code/blob/master/.tool-versions#L1

If you use https://asdf-vm.com/, it will use that file to ensure that you are using the correct node version.

dlech, 2023-02-13:
is this still a problem?

laurensvalk, 2023-02-14:
I haven't run Pybricks Code locally much lately, so let's close it until it causes any problems.

closed 2023-02-14 by laurensvalk