pybricks/support #873: [Bug] Import does not work in V3.2 release for some scripts

laurensvalk, 2022-12-22:
**Describe the bug**
Importing does not work in some cases. I found a reproducible case that appears to hinge on `async` appearing somewhere in the code, even if it isn't used.

The following reproduces it always in Pybricks Code:

*my_module.py*
```python
data = 123
```

*my_main.py*
```python
from my_module import data

async def hello():
    print("hello")

print(data)
```

Running the latter gives

```
ImportError: no module named 'my_module'
```

dlech, 2022-12-22:
Any errors in the browser console? Does it work with pybricksdev?

laurensvalk, 2022-12-22:
No errors in console. Pybricksdev test is scheduled for Friday.

dlech, 2022-12-22:
Apparently errors for this are only enabled in development builds.  I get this error in the console:
```
Error: Parse error on line 3:
... import dataasync def hello():    pri
---------------------^
Expecting 'NEWLINE', '(', ')', ':', ',', '=', '*', '**', ';', '+=', '-=', '*=', '/=', '%=', '&=', '|=', '^=', '<<=', '>>=', '**=', '//=', 'from', '.', 'as', 'if', 'else', 'for', 'in', 'or', 'and', 'not', '<', '>', '==', '>=', '<=', '!=', 'is', '|', '^', '&', '<<', '>>', '+', '-', '/', '%', '//', '[', ']', '}', got 'def'
    at Parser.parseError (python3.js:603:1)
    at Parser.parse (python3.js:670:1)
    at exports.parse (python3.js:1256:1)
    at parse (python-parser.ts:30:1)
    at findImportedModules (lib.ts:86:1)
    at handleCompileMulti6 (sagas.ts:151:1)
    at handleCompileMulti6.next (<anonymous>)
    at next (redux-saga-core.esm.js:1157:1)
    at currCb (redux-saga-core.esm.js:1251:1)
    at takeCb (redux-saga-core.esm.js:503:1)
```

I guess the parsing library we are using has a problem with async syntax?

closed 2022-12-22 by dlech

dlech, 2022-12-22:
It turns out the parser library we are using in Pybricks Code was developed against Python 3.4, so any newer Python features supported by MicroPython need to be added.