pybricks/support #874: enable more builtins

dlech, 2022-12-23:
I was coding today and wanted to use `filter` but it is not enabled. 

We've also considered enabling `set` in the past.

For people writing classes, they may also expect `property` to be enabled.

`memoryview` and `frozenset` are also currently disabled, but I can't think of any compelling use case for enabling them.

dlech, 2022-12-27:
The 2-arg version of `next()` is potentially useful for parallel tasks using generators.

dlech, 2023-01-23:
The `gc` module may be potentially useful for getting more constant loop times.

apumapho, 2023-04-30:
@dlech, +1 for `set` being somewhat valuable for use cases where people are trying to maintain unique `list`s of things and not having to maintain uniqueness themselves