pybricks/support #877: Motor.track_target does not work as expected

dlech, 2022-12-23:
**Describe the bug**
I'm trying to use `Motor.track_target()` to make Gelo walk by running motors 180 degrees out of phase.

If there is a 10 ms delay between calls to `track_target`, it seems to work as expected, but if there is only a 1 ms delay, the motors  all get in phase after a short time instead of staying 180 degrees out of phase.

**To reproduce**

Firmware v3.2.0.

Steps to reproduce the behavior:
1. Build the Gelo model with the Robot Inventor kit
2. Run the program below.

**Expected behavior**
The position of the opposing motors should stay 180 degrees out of phase.

**Screenshots**
In this video, you can see that the Gelo start out walking forward as expected, but the motors quickly go in phase and it starts going backwards instead.

https://user-images.githubusercontent.com/963645/209411733-df7c35a1-cd1c-40fb-8cf7-4d58b0572042.mp4

You can also see this in the printed output, at the beginning, two motors are near 0 and two are near 180 and they stay that far apart.

```
0 [7, 182, 180, 3]
45 [7, 186, 183, 3]
90 [19, 200, 196, 9]
135 [45, 216, 220, 30]
180 [79, 248, 251, 58]
225 [115, 274, 284, 95]
270 [155, 298, 320, 134]
```

But by the end, they are all over the place.
```
6345 [56, 17, 237, 258]
6390 [96, 64, 273, 298]
6435 [138, 101, 313, 338]
6480 [176, 139, 345, 17]
6525 [212, 179, 24, 65]
6570 [250, 213, 68, 106]
6615 [286, 254, 113, 147]
```

**Script**
```python
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

from lib_dataclass import dataclass

CW = Direction.CLOCKWISE
CCW = Direction.COUNTERCLOCKWISE
OPP_ANGLE = 180

hub = InventorHub()

back_right = Motor(Port.A, CCW)
back_left = Motor(Port.B, CW)
front_right = Motor(Port.C, CCW)
front_left = Motor(Port.D, CW)
color = ColorSensor(Port.F)
ultrasonic = UltrasonicSensor(Port.E)

all_legs = [back_right, back_left, front_right, front_left]

# Motor.__hash__ is broken
# https://github.com/pybricks/support/issues/876
# opposition = {
#     back_right: 0,
#     back_left: 0,
#     front_right: 0,
#     front_left: 0,
# }

def opposition(leg):
    if leg is back_right:
        return 0
    if leg is back_left:
        return OPP_ANGLE
    if leg is front_right:
        return OPP_ANGLE
    if leg is front_left:
        return 0

# Ensure legs travel shortest distance to reach opposition after init.
# Often, the legs will coast past 180 when stopping so when the program
# restarts, they come up as -176 and would have to rotate nearly 360 degrees
# to reach opposition.
for leg in all_legs:
    if opposition(leg) - leg.angle() > OPP_ANGLE:
        leg.reset_angle(leg.angle() + 360)

def main():
    oppose_legs()     

    for t in range(360 * 50):
        track_target(t)
        if t % 45 == 0:
            print(t, [leg.angle() % 360 for leg in all_legs])
        wait(1)

    wait(1000)
    

def track_target(target):
    for leg in all_legs:
        leg.track_target(target + opposition(leg))

def oppose_legs():
    """
    Sets legs to opposing positions.
    """
    for leg in all_legs:
        leg.run_target(400, opposition(leg), wait=False)

    while not all(map(Motor.done, all_legs)):
        wait(10)

def zero_legs():
    """
    Sets all legs to the "zero" position.
    """
    for leg in all_legs:
        leg.run_target(400, 0, wait=False)

    while not all(map(Motor.done, all_legs)):
        wait(10)

def beep(down):
    beeps = [100, 200, 300]
    
    if down:
        beeps = reversed(beeps)

    for b in beeps:
        hub.speaker.beep(b)

try:
    # https://github.com/pybricks/support/issues/874
    filter
except NameError:
    def filter(func, items):
        for i in items:
            if func(i):
                yield i

print(f"battery: {hub.battery.voltage() / 1000:.2f}V")
color.lights.off()
ultrasonic.lights.on(10)
beep(False)

try:
    main()
finally:
    ultrasonic.lights.off()
    color.lights.off()
    beep(True)

```

dlech, 2022-12-23:
I forgot to do my math. 1 degree per millisecond is way faster than the motors can turn.

dlech, 2022-12-23:
For a minute, I thought my math was off, but this should be 1000 degrees per second, which should be within the capabilities of the motor. Maybe I'm missing something?

dlech, 2022-12-23:
I changed the print line to

```python
            print(t, [leg.angle() - opposition(leg) for leg in all_legs])
```

and got 

```
1665 [1309, 1221, 1289, 1419]
1710 [1343, 1254, 1334, 1462]
1755 [1376, 1291, 1379, 1503]
```

So it does seem that the motors are indeed falling far behind the target.

dlech, 2022-12-23:
```
>>> m.control.limits()
(1080, 2000, 199)
```

This says max speed for these motors is 1080 deg/sec.

dlech, 2022-12-23:
I changed to a time-based loop to ensure an exact speed.

```python
    while timer.time() < 60000:
        t = 800 * timer.time() // 1000
        track_target(t)
        if t % 360 == 0:
            print(t, [leg.angle() - opposition(leg) for leg in all_legs])
        wait(10)
```

It seems to do OK at 800 degrees per second but falls behind at 900 degrees per second.

laurensvalk, 2022-12-24:
Ha, this time it's my turn to say it. Can we have logs, please? :smile: 

We have an existing test (`tests/motors/loop_time.py`) that is somewhat similar. Perhaps you can modify it to reproduce the problem you found.

Just hitting F5 with the `Motor Test` launch configuration will run your script, retrieve the logs, and make a graph.

You'll see an interactive graph pop up, but you'll also find static PNG files in `tests/motors/build/....`

[image]

dlech, 2022-12-24:
> Can we have logs, please?

Sure thing. But to be clear, I don't think there is any bug here. I just forgot my lessons from ev3dev. So it's really a matter of figuring out the best way to run 4 motors as fast as possible while staying in sync.

The lesson I forgot is that the actual max speed of the motor depends on the battery voltage and the load on the motor. The max speed reported in the limits is probably a no-load speed at some certain voltage (in ev3dev, I think we defined max speed as no-load speed at 9V).

On the Gelo model, there is variable load on all of the motors while it is walking and the net average load is random since the robot bounces around quite a bit and can hit obstacles, etc.

So what I'm looking for is the best way to get feedback that `track_target()` is maxed out and can't keep up so that I can limit the requested target accordingly.

dlech, 2022-12-24:
Modified the loop in the test to:

```python
while watch.time() < DURATION + 500:
    motor.track_target(1000 * watch.time() // 1000)
```

Here are the logs:

[servo.txt](https://github.com/pybricks/support/files/10299021/servo.txt)
[control.txt](https://github.com/pybricks/support/files/10299022/control.txt)
[hub_output.txt](https://github.com/pybricks/support/files/10299023/hub_output.txt)
[image]
[image]

dlech, 2022-12-24:
As expected, the graphs show the reference increasing faster than the motor can physically keep up so the error keeps increasing too while the voltage is maxed out.

laurensvalk, 2022-12-24:
> So what I'm looking for is the best way to get feedback that track_target() is maxed out and can't keep up so that I can limit the requested target accordingly.

> 
```
>>> m.control.limits()
(1080, 2000, 199)
```

The second value is the torque limit. You can see it max out to this value in the graph.

You can read `Motor.load()`, which should give this value dynamically. You could throttle back the reference speed if it gets closer to the torque limit.

Throttling might work best on a slow moving average basis (initialized at a reasonable value) so that the throttling itself does not cause additional feedback loops at similar frequencies of walking.

dlech, 2022-12-24:
Thanks for the tip. This is what I came up with:

```python
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

_CW = Direction.CLOCKWISE
_CCW = Direction.COUNTERCLOCKWISE

# index to limits() tuple
_TORQUE = const(2)

_BEEP_START = const(1)
_BEEP_END = const(2)
_BEEP_ERROR = const(3)

def _copy_sign(x, y):
    """
    Like math.copysign(), but for integers.
    """
    if y < 0:
        return -x

    return x

class Gelo:
    def __init__(self):
        self.hub = InventorHub()

        self.back_right = Motor(Port.A, _CCW)
        self.back_left = Motor(Port.B, _CW)
        self.front_right = Motor(Port.C, _CCW)
        self.front_left = Motor(Port.D, _CW)
        self.color = ColorSensor(Port.F)
        self.ultrasonic = UltrasonicSensor(Port.E)

        self.all_legs = [
            self.back_right,
            self.back_left,
            self.front_right,
            self.front_left,
        ]

        self.near_max_load = 90 * sum([
            l.control.limits()[_TORQUE] for l in self.all_legs
        ]) // 100

        # Motor.__hash__ is broken
        # https://github.com/pybricks/support/issues/876
        # self._opposition = {
        #     back_right: 0,
        #     back_left: 180,
        #     front_right: 180,
        #     front_left: 0,
        # }

    def __enter__(self):
        self.color.lights.off()
        self.ultrasonic.lights.on(10)
        self._beep(_BEEP_START)

        return self

    def __exit__(self, exc, value, trace):
        for leg in self.all_legs:
            leg.stop()

        self.ultrasonic.lights.off()
        self.color.lights.off()

        wait(500)

        if exc:
            self._beep(_BEEP_ERROR)
        else:
            self._beep(_BEEP_END)

    def _beep(self, kind):
        if kind == _BEEP_ERROR:
            for b in [100, 50]:
                self.hub.speaker.beep(b, 300)
        else:
            beeps = [100, 200, 300]
        
            if kind == _BEEP_END:
                beeps = reversed(beeps)

            for b in beeps:
                self.hub.speaker.beep(b)

    def _opposition(self, leg):
        if leg is self.back_right:
            return 0
        if leg is self.back_left:
            return 180
        if leg is self.front_right:
            return 180
        if leg is self.front_left:
            return 0

    def _reset_legs(self):
        """
        Resets all legs to the absolute position as measured by the motor.
        """

        for leg in self.all_legs:
            leg.reset_angle()
        
            # Ensure legs travel shortest distance to reach opposition after
            # init. Often, the legs will coast past 180 when stopping so when
            # the program restarts, they come up with and angle of -176 degrees,
            # for example, and would have to rotate nearly 360 degrees to reach
            # opposition.
            if self._opposition(leg) - leg.angle() > 180:
                leg.reset_angle(leg.angle() + 360)

    def _track_target(self, target):
        for leg in self.all_legs:
            leg.track_target(target + self._opposition(leg))

    def oppose_legs(self):
        """
        Sets legs to opposing positions.
        """
        self._reset_legs()

        for leg in self.all_legs:
            leg.run_target(400, self._opposition(leg), wait=False)

        while not all(map(Motor.done, self.all_legs)):
            wait(10)

    def zero_legs(self):
        """
        Sets all legs to the "zero" position.
        """
        self._reset_legs()
        
        for leg in self.all_legs:
            leg.run_target(400, 0, wait=False)

        while not all(map(Motor.done, self.all_legs)):
            wait(10)

    def walk(self, speed, duration):
        self._reset_legs()

        # The actual fastest rate the motors can turn on
        # Gelo is somewhere between 700 and 900 deg/sec
        # depedning on battery voltage and terrain.
        # We will adjust the rate down if needed.    
        target_rate = max(-900, min(speed, 900))

        timer = StopWatch()

        while timer.time() < duration:
            t = target_rate * timer.time() // 1000
            self._track_target(t)

            load = sum(map(Motor.load, self.all_legs))

            # If we are trying to go faster than the motors
            # can actually turn, we need reduce the rate so
            # that they can keep up.
            if abs(load) > self.near_max_load:
                target_rate -= _copy_sign(10, target_rate)

            wait(10)

    def print_battery_voltage(self):
        print(f"battery: {self.hub.battery.voltage() / 1000:.2f}V")

if __name__ == "__main__":
    with Gelo() as gelo:
        gelo.print_battery_voltage()
        gelo.walk(900, 5000)   
        gelo.walk(-900, 5000)   
```

closed 2023-01-06 by dlech