pybricks/support #883: Use named tuples for limits and settings.

dlech, 2022-12-24:
**Is your feature request related to a problem? Please describe.**
It is not very clear that `Motor.control.limits()[2]` is the `torque` limit.

**Describe the solution you'd like**
If we create a named tuple for this, we could write `Motor.control.limits().torque` instead (without breaking existing code).

**Describe alternatives you've considered**
As a workaround, we can create a local constant for the index.

```python
TORQUE = 2

m = Motor(...)
print(m.control.limits()[TORQUE]
````

**Additional context**
We could make this optional and exclude it on move hub to save firmware size.

laurensvalk, 2022-12-24:
The recommended way to read these settings currently is:

```python
speed_limit, acceleration, torque_limit = motor.control.limits()
```

dlech, 2022-12-24:
That doesn't really work for cases like this:

```python
sum([m.control.limits()[2] for m in motors])
```

laurensvalk, 2022-12-24:
In regards to the recommended usage, perhaps we can make the signatures clearer like this.

Current:

[image]

idea:

[image]

laurensvalk, 2022-12-24:
> That doesn't really work for cases like this though:
> 
> ```python
> sum([m.control.limits()[2] for m in motors])
> ```

I would argue this to be a fairly advanced use case. The main uses of these methods is setting values to improve defaults.

closed 2023-02-17 by dlech