pybricks/support #885: [Bug] imu.angular_velocity() returns accelerometer values rather than gyroscope values (Technic Hub)

CasperClarke, 2022-12-25:
**Describe the bug**
When using imu.angular_velocity() on a technic hub accelerometer values are returned rather than gyroscope values.

**To reproduce**
In the Pybricks online IDE type the following code;

```
from pybricks.hubs import TechnicHub
from pybricks.tools import wait
from pybricks.geometry import Axis

# Initialize the hub.
hub = TechnicHub()

# Get the acceleration or angular_velocity along a single axis.
# If you need only one value, this is more memory efficient.
while True:

    # Read the forward acceleration.
    #forward_acceleration = hub.imu.acceleration(Axis.X)

    # Read the yaw rate.
    G_rateX = hub.imu.angular_velocity(Axis.X)
    G_rateY = hub.imu.angular_velocity(Axis.Y)
    G_rateZ = hub.imu.angular_velocity(Axis.Z)
    A_rateX = hub.imu.acceleration(Axis.X)
    A_rateY = hub.imu.acceleration(Axis.Y)
    A_rateZ = hub.imu.acceleration(Axis.Z)

    # Print the yaw rate.
    print(" Gyro: ",yaw_rateX,yaw_rateY,yaw_rateZ)
    print("Accel: ",A_rateX,A_rateY,A_rateZ)
    wait(100)
```
Produces the result:
```
 Gyro:  1826.347 -3906.421 9126.949
Accel:  1826.347 -3906.421 9126.949
 Gyro:  1831.135 -3915.995 9047.959
Accel:  1831.135 -3915.995 9047.959
```

**Expected behavior**
The expected result would be an angular velocity in deg/s would be returned from the lsm6dso in the form of 3 values along each axis.

CasperClarke, 2022-12-26:
I think I have found the source of the error.

The code found here:
https://github.com/pybricks/pybricks-micropython/blob/af76eecd6bfaff97c9e6435ed16d4bc97b9a946e/lib/pbio/drv/imu/imu_lsm6ds3tr_c_stm32.c#L283-L298
is sourcing both the accel and gyro values from the same source.

To fix the problem the code should be changed to the following for the gryo read function:
```
void pbdrv_imu_gyro_read(pbdrv_imu_dev_t *imu_dev, float *values) {
    // Output is signed such that we have a right handed coordinate system
    // consistent with the coordinate system above. Positive rotations along
    // those axes then follow the right hand rule.
    values[0] = PBDRV_CONFIG_IMU_LSM6S3TR_C_STM32_SIGN_X * imu_dev->data[3] * imu_dev->gyro_scale;
    values[1] = PBDRV_CONFIG_IMU_LSM6S3TR_C_STM32_SIGN_Y * imu_dev->data[4] * imu_dev->gyro_scale;
    values[2] = PBDRV_CONFIG_IMU_LSM6S3TR_C_STM32_SIGN_Z * imu_dev->data[5] * imu_dev->gyro_scale;
}
```
The original code was using the accel scaling and the data from the accelerometer, indices 0-2, while the gyro is stored in indices 3-5 as notated here:
https://github.com/pybricks/pybricks-micropython/blob/af76eecd6bfaff97c9e6435ed16d4bc97b9a946e/lib/pbio/drv/imu/imu_lsm6ds3tr_c_stm32.c#L240-L246

I don't know how to compile the code to create a custom firmware in order to test it, so I cannot confirm the solution works at this time.

CasperClarke, 2022-12-26:
I have now compiled and tested a local build of the code and submitted a pull request to fix the issue for [pybricks-micropython](https://github.com/pybricks/pybricks-micropython). I have tested the code on the lego technic hub and it worked successfully.

laurensvalk, 2022-12-26:
Thanks a lot for reporting, investigating, and even fixing the issue! We really appreciate it. I'll be merging your pull requets as soon as possible.

closed 2022-12-26 by laurensvalk