pybricks/support #894: [Bug] Code autocomplete pastes "abs" after Space then Enter

davecparker, 2022-12-30:
Code auto-completion gives a long list of global symbols after you type a space in an expression or unfinished statement (which is questionable, but I'm not flagging that here), and if you press Enter because you intend to finish that line later or continue it on the next line, it pastes "abs". This even happens when you are typing a comment (where space at end of a line that needs to be continued is a normal habit).

**To reproduce**
Steps to reproduce the behavior:
1. Start a line of code that says "x = "  (with the trailing space)
2. Press Enter
3. "abs" is pasted

**Suggestion**
If you don't have another way to prevent this, could you make the first entry in the global name list blank (i.e. none)?

dlech, 2022-12-30:
How about just disabling the trigger on space?

dlech, 2022-12-30:
I dropped the trigger on space. You can try it at https://beta.pybricks.com. (v2.1.0-beta.4)

There is still a trigger on `.` which can be annoying on comments, but if we remove it, it means you have to press <kbd>ctrl</kbd>+<kbd>space</kbd> after every dot to get completion.

laurensvalk, 2023-01-12:
> I dropped the trigger on space.

Is there a way to suppress/enable it for specific cases? Now it suppresses import suggestions, e.g.

```python
from pybricks.parameters import       # <--- now we don't get suggestions here
```

dlech, 2023-02-13:
Practically speaking, I don't think we can improve this any further without replacing/forking both the monaco editor and jedi, so I think we should close the issue.

closed 2023-03-07 by dlech