pybricks/support #896: [Bug] Documentation won't load in Pybricks Beta

laurensvalk, 2023-01-02:
This seems to happen only in my offline version of Pybricks Beta. If I open it in a new profile, the docs do load.

A hard refresh does not seem to help.

[image]

dlech, 2023-01-02:
In the chrome dev tools, go to the *Application* tab, select *Service Workers* and click *Unregister*, then refresh the page. 

[image]

There was an older release that had a bug like this, so if you were upgrading from that version, it could explain the issue.

dlech, 2023-01-02:
Part of the problem is that chrome doesn't clear the cache for the iframe when you do <kbd>ctrl</kbd>+<kbd>r</kbd>.

dlech, 2023-02-13:
is this still a problem?

laurensvalk, 2023-02-14:
Yes. But I haven't cleared the cache yet as I don't want to manually re-upload a hundred programs.

I was just going to wait until the next release to see if that refreshes the docs.

dlech, 2023-02-14:
There is nothing a new release can do to fix the problem since the bug was caused by a previous release. Clearing the cache and unregistering the app does not clear the indexeddb where the files are stored.