pybricks/support #899: Technic Hub refuses to load pybricks firmware and shows error

MCreeper1, 2023-01-03:
### Discussed in https://github.com/orgs/pybricks/discussions/898

<div type='discussions-op-text'>

<sup>Originally posted by **MCreeper1** January  3, 2023</sup>
I have tried multiple times to load the pybricks firmware onto the hub and I always get the same error screen during the installation process. After I pair the hub goes from purple to rainbow then after about 30 seconds it stops and starts flashing white.A error screen appear that  says :
```
Error: Failed to execute 'getPrimaryService' on 'BluetoothRemoteGATTServer': GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
![New tab and 4 more pages - Personal - Microsoft​ Edge 1_3_2023 4_12_03 PM](https://user-images.githubusercontent.com/121881087/210385731-65f0be0b-d67b-4ed7-8f57-930c086ed323.png)

```
I have tried switching from chrome to edge but the result is the same.</div>

dlech, 2023-01-03:
Since you said you tried Edge, is it safe to assume you are using Windows? Unfortunately, lot's of people have trouble with Bluetooth on Windows. So this is probably the same issue as #312, #513, #555, etc.

It would be useful to know what Bluetooth chip/driver you are using. Run this in PowerShell to find out:

```powershell
wget https://github.com/Microsoft/busiotools/raw/master/bluetooth/tracing/GetBluetoothRadioInfo.ps1 -UseBasicParsing | iex
```

You could also try logging Bluetooth packets as described in https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

closed 2023-01-04 by dlech

dlech, 2023-01-06:
Thanks for the logs. This has the same "LE Read Remote Version Information" response timeout as in #513. There are not any known solutions, but make sure you hub has new/fully charged batteries. You could try flashing the firmware with another device, like an Android phone. You could try eliminating any sources of interference, like Wi-Fi and other Bluetooth devices. You could try a different Bluetooth dongle or try different driver versions for your Bluetooth adapter.

NM230105, 2023-01-06:
Thanks for the quick reply! We'll give these a try.

NM230105, 2023-01-21:
Just thought I’d send an update. I bought a $15 tp-link UB500 Bluetooth dongle (BT v5.0) for the Win11 box and that did the trick. 

From the specs I uploaded, the original Bluetooth adapter was v4.1 with its smaller packet size and given the timeout errors it’s likely that was the root of the problem. v5.0 definitely works and I suspect that even v4.2 would have done the job as that was the first version with the increased packet size limit hence 2X faster data throughput.

dlech, 2023-01-21:
Thanks for the followup. Would you mind running this (in powershell) and sharing the results so we can get an idea of which adapters/drivers work and which don't?

    wget https://github.com/Microsoft/busiotools/raw/master/bluetooth/tracing/GetBluetoothRadioInfo.ps1 -UseBasicParsing | iex