pybricks/support #904: Add close() method to Motor classes

dlech, 2023-01-06:
I have been trying to cope with this situation but it is impossible for my program. The whole ideia of Remote Bla Bla to be able to configure the devices I want to use by trial and error. That means configure, test, reconfigure. For that I need to connect and reconnect devices.

Can you please consider making a Motor/DCMotor.close() method as soon as possible? Beta is ok for me... Thank you!

_Originally posted by @vascolp in https://github.com/pybricks/support/discussions/888#discussioncomment-4575081_

laurensvalk, 2023-01-10:
Well, we did it :smile: . Here's a firmware version that has this. Do you want to try and test it?

https://github.com/pybricks/pybricks-micropython/suites/10284401577/artifacts/505219513

vascolp, 2023-01-10:
Wow that was fast!
I would love to try it now ... but I have to work...
Give me a couple of days days.
Thank you!

laurensvalk, 2023-01-10:
Note to ourselves: We might need to rethink this implementation if we used the motor itself as a generator object.

vascolp, 2023-01-13:
Approved!

laurensvalk, 2023-06-08:
This seems to have been accidentally deleted when revisiting async. It should not interfere with the way we ended up using async, so we should be able to bring it back.

vascolp, 2023-06-08:
Thank you! 
I noticed it wasn´t available in the latest beta... the one with the observe_channels which I want to test!

laurensvalk, 2023-06-08:
We've been making huge changes (and improvements) under the hood. If this is the worst of it, we haven't done so bad :smile:

closed 2023-06-09 by laurensvalk

laurensvalk, 2023-06-09:
Should be fixed now. It will be available in the next beta. If you'd like to try it sooner, just grab the latest firmware from:

https://nightly.link/pybricks/pybricks-micropython/workflows/build/master