pybricks/support #905: Robot Inventor Hub refuses to connect to pybricks app to install firmware

ODog2028, 2023-01-07:
I am trying to install the pybricks firmware on my robot inventor hub but I get an error when connecting the hub to the software. The error I get is:
```
Error: Failed to execute 'open' on 'USBDevice': Access denied.
```

dlech, 2023-01-07:
What operating system?

dlech, 2023-01-07:
How are you trying to flash the firmware?

ODog2028, 2023-01-07:
I am using ChromeOS and I have tried from both code.pybricks.com and dfu.pybricks.com but got the same error.

ODog2028, 2023-01-07:
I have figured out the issue and I uploaded the firmware!

closed 2023-01-07 by ODog2028

laurensvalk, 2023-01-07:
Glad to hear it worked. What was the issue?

ODog2028, 2023-01-07:
ChromeOS was trying to give access to the hub to android and chrome at the same time.

laurensvalk, 2023-01-07:
Thanks! That helps clear it up in case anyone with the same issue finds this later.

Have fun coding now :)