pybricks/support #911: [Bug] Starting a builtin program after opening connection dialog locks Pybricks Code connect button

laurensvalk, 2023-01-10:
**Describe the bug**
Starting a builtin program after opening connection dialog locks Pybricks Code connect button

**To reproduce (prepare)**
- Save this program on the hub (Prime/Technic).

```python
from pybricks.tools import wait

while True:
    wait(1000)
```

- In practice this can happen with any infinitely running program already stored on the hub. This is just a simple test case.

**To reproduce**
- Click Bluetooth icon to open connection dialog
- Start the program on the hub with the button
- Attempt to connect because you can still see the hub listed (which we can't work around)
- Pybricks Code connection spinner runs indefinitely.

**Expected behavior**
Connection failure?

dlech, 2023-01-10:
This looks like a Chrome bug. I filed https://crbug.com/1406311.

A simpler way to reproduce is to just turn off the hub instead of starting a program.

I tried adding a timeout but that just made things worse. Since it is still waiting for the previous connect, you can't open a new scan dialog.

dlech, 2023-01-11:
Apparently the Chrome bug has been open since 2017, so this might be one we have to fix ourselves upstream. https://crbug.com/684073