pybricks/support #913: [Bug] PoweredUp Light (item #88005) uncontrollable on Move Hub (item #88006)

StuffAndyMakes, 2023-01-10:
**Describe the bug**
Running "Making the light blink" example from the [Light](https://docs.pybricks.com/en/stable/pupdevices/light.html) object documentation connected to `Port.C` on a Move Hub results in the motor on `Port.A` starting, stopping, starting, stopping, etc. ("blinking"). 

**To reproduce**
Run the following example code (modified to use `Port.C` for the LED module):

```
from pybricks.pupdevices import Light
from pybricks.parameters import Port
from pybricks.tools import wait

# Initialize the light.
light = Light(Port.C)

# Blink the light forever.
while True:
    # Turn the light on at 100% brightness.
    light.on(100)
    wait(500)

    # Turn the light off.
    light.off()
    wait(500)
```

**Expected behavior**
The expected behavior is for the LEDs on the 88005 PoweredUp Light to blink on and off.

**Screenshots**
https://user-images.githubusercontent.com/2899323/211683271-866a9e6f-c2f6-445b-8875-33f472988464.mov

dlech, 2023-01-11:
Firmware version?

dlech, 2023-01-11:
I'm able to reproduce with v3.2.2.

dlech, 2023-01-11:
Thanks for the report. Who knows how long it has been like that. I guess not many people are using the light on the move hub. :smile: 

You can grab a firmware with the fix from https://github.com/pybricks/pybricks-micropython/suites/10297481669/artifacts/506150765 if you don't want to wait for the next release.

closed 2023-01-11 by dlech

yobsh, 2023-02-17:
how do you install that firmware? or what do you do with that zip file?

dlech, 2023-02-17:
> how do you install that firmware? or what do you do with that zip file?

Please see https://pybricks.com/install/technic-boost-city/#installing-the-latest-build-advanced

StuffAndyMakes, 2023-02-17:
Thanks for the quick turnaround! Love it!

yobsh, 2023-02-18:
it works awesome! thanks!