pybricks/support #916: [Bug]

TortugaSlim, 2023-01-14:
**Describe the bug**
Using the remote to control the Transformation Vehicle (42140)

initially I was using the print function to print values to the screen for troubleshooting, while it got things working.

Basic functionality:
push LEFT_PLUS -> drive forward
release LEFT_PLUS -> stop
replicated for forwards, reverse, left, right etc...

Once I had the functionality I desired, I commented out the Print lines. 

Then the code started missbehaving. Press a button on the remote and the car would start to drive, but releasing the button would not stop the car, it just kept going.

I tried several different strategies to stop the car when no buttons were pressed, but none of them would work. Then I realised the Print command was actually significantly slowing down the cycle time of the program. Polling the remote and them immediatly using the print command was effectively adding a wait() into the system...

I threw a wait(100) in right after polling the remote and the problem went away... i then reduced that to 10ms and it still works.
I haven't attempted to reduce this as 10ms is hardly noticable when driving the vehicle.

but when I take the delay out, the problem returns.

**To reproduce**
Steps to reproduce the behavior:
1. open code attached
2. attach motors to ports A and B on a TechnicHub
3. run code
4.  press button
5. observe motors start
6. release button
7. observe mototrs stop
8. comment out line 36
9. run code
10. press button
11. observe motors start
12. release button
13. observe motors continue running
14. uncomment line 40 or 41
15. run code
16. press button
17. observe motors start
18. release button
19. observe motors stop

**Expected behavior**
More something I did not expect. I did not expect that commenting out the PRINT commnad would cause the remote buttons to be read incorrectly.

Side Note: htis may be another issue all together, but polling the remote, immediatly followed by the print command occasionally led to problems executing line 35. the the program would error out. I did not take a shot of the error code (appologies). But commenting out the print commnad has fixed the issue. You may experience this as well while following the steps above, but it was intermittant.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
[tumbler.txt](https://github.com/pybricks/support/files/10417091/tumbler.txt)

laurensvalk, 2023-01-14:
Thanks for reporting!

Does this problem go away if you move the line:

```python
my_remote.light.on(Color.BLUE)
```
out of the loop? For example place it just before it.

TortugaSlim, 2023-01-14:
now I feel foolish, I can't reproduce it!

I removed the wait(10) and it's fine..

i tried different combinations with the remote.light.on(Color.Blue) commented out and included. it worked fine each time.

can't reproduce the behaviour... very strange, but all good I suppose!

thanks for your help!

dlech, 2023-03-20:
The communication protocol between the remote and the hub is lossy (technically speaking, it is using GATT notifications which have no guarantee of delivery). So although it is quire reliable in practice, it is can explain the observed but not reproducible behavior.

So I'm going to go ahead and close the issue since we don't have a reproducible case. We can always open it again if we get new information.

closed 2023-03-20 by dlech