pybricks/support #917: [Feature] Multi-tasking ideas and working example approach

laurensvalk, 2023-01-14:
**Introduction: Why multi-tasking?**
This is a common feature request. When people graduate from block-coding languages to text-based languages, parallelism is suddenly much harder to do. It is technically already possible in Pybricks (see https://github.com/orgs/pybricks/discussions/356 and https://github.com/pybricks/support/issues/357), but it is not trivial.

This post discusses some ideas of how we might make it more accessible. Note that there are many possible approaches, so this might not make it into the final version. We welcome ideas and suggestions on this topic.

**An example**
Here is an example of running two or more "tasks" in parallel, which is what most feature requests have been about so far.

You can already run this on the hub (see instructions below). It is quite fun to play with.

```python
from pybricks.hubs import ThisHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Direction, Stop, Color
from pybricks.robotics import DriveBase
from pybricks.tools import wait

# Experimental module for multi-tasking. Importing turns all time-consuming
# operations like motor movement into awaitables so they can run in parallel.
import task 

# Initialize devices as usual.
hub = ThisHub()
left_motor = Motor(Port.A, Direction.COUNTERCLOCKWISE)
right_motor = Motor(Port.B)
drive_base = DriveBase(left_motor, right_motor, wheel_diameter=56, axle_track=112)

# Task to drive in a square.
async def square():
    for side in range(4):
        # Drive forward and then turn.
        await drive_base.straight(200)
        await drive_base.turn(90)

# Task to blink.
async def blink():
    for i in range(10):
        hub.light.on(Color.RED)
        await wait(500)
        hub.light.on(Color.GREEN)
        await wait(500)

# Hello world task.
async def hello(name):
    print("Hello!")
    await wait(2000)
    print(name)

# This is the main program
@task.main
async def main():
    print("Running multiple tasks at once!")
    await task.run_parallel(square(), blink(), hello("Pybricks"))
    print("You can also just run one task.")
    await hello("World!")
    print("Done!")
```
**How does this work?**
All operations that take some time to complete normally block the rest of your program. For example, this includes `wait`, `hub.speaker.beep`, `motor.run_angle(500, 90)`, or `drive_base.turn(90)`.

When the (experimental) `task` module is imported, a few things change in the Pybricks modules. All of these normally blocking operations now allow other things to run in parallel!

To do this, functions (`def`) are written as `async def`, and any normally blocking operation is preceded by `await`. This tells your hub that it can execute other things in the mean time and come back when the operation is complete.

**You can try this example!**
- Sign into GitHub and go [here](https://github.com/pybricks/pybricks-micropython/actions/runs/3918707616)
- Scroll to the bottom of the page and download the firmware file for your hub.
- In the [Pybricks Code](https://code.pybricks.com/) settings menu, click `Install Pybricks Firmware` as usual.
- Instead of selecting a hub, click `Advanced` and follow the instructions to pick your downloaded file.
- Proceed as usual. Now run the program above on any hub.

**This example isn't entirely correct Python**
All of the above is internally implemented using generators as in https://github.com/pybricks/support/issues/357. Instead of using `async/await`, we should therefore technically omit `async` and replace `await` with `yield from` everywhere. It currently just works with `await` due to how MicroPython works. There are, however, other ways to do the internal implementation so that it could still be correct Python.

Rather, the reason for using `await` here instead of `yield from` is because it is simpler for new users. This is also commonly used in "proper" cooperative multi-tasking with asyncio.

**A controversial thought experiment**
If you look at the example, you could think about leaving out `async` and `await` entirely from the user code. These keywords could be inserted behind the scenes when downloading the program.

Then it looks just the same as today's code, except you can now do things in parallel.

The controversial part is that we'd no longer truly be using Python/MicroPython, which could confuse advanced users.

On the other hand, it would resolve the non-Pythonicness above. `async` could be left out together, and we'd just be inserting `yield from` in front of every "awaitable" if it is missing. This allows advanced users to still write correct code with generators and beginning users wouldn't have to worry about it.

dlech, 2023-01-14:
This was keeping me awake last night. :stuck_out_tongue: I had a few useful thoughts that are worth writing down.

### coroutines vs. generators

Here is a simple program that shows the different between async/await in Python vs MicroPython:

```python
async def test():
    ...

print(type(test()))
```

And the output

```
david@computer:~/work/pybricks/micropython$ ./ports/unix/build-standard/micropython test.py 
<class 'generator'>
david@computer:~/work/pybricks/micropython$ python3 test.py
/home/david/work/pybricks/micropython/test.py:4: RuntimeWarning: coroutine 'test' was never awaited
  print(type(test()))
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
<class 'coroutine'>
```

In MicroPython, `async`/`await` is just syntactic sugar for "this is a generator"/`yield from`. `async` in `async def` is mostly ignored other than to flag a function as a generator and allow `async with` and `async for` statements and `await` is compiled as if it was `yield from`. Therefore, calling the method just returns a generator object.

In Python, `async def` causes a method to be compiled to a `coroutine` object instead of a `generator` object. Coroutines have a helpful feature to catch the case where you forget to `await`. When the object is garbage collected, it raises a `RuntimeWarning` to let you know this happened and where the exact line of code is that needs to be fixed.

If we want to catch not awaited async tasks at runtime, this seems like the "correct" way to do it. To make it work in MicroPython in general, it would likely require modifying the compiler and runtime to differentiate between coroutines and generators. For the `pybricks` APIs we could probably implement something like this without such modifications. Also, since this would be trigger by garbage collection, we would likely want to run the gc on every iteration of the main loop so that we raise the warning/exception asap (running the gc like this could have other benefits as well, like helping to avoid fragmentation and keeping the loop time more constant).

### a redux-sagas inspired async library

In Pybricks Code we use https://redux-saga.js.org. Of all of the async frameworks I have used, I think this one has the most useful constructs for the type of program needed for robotics.

In Python, `asyncio` is the "batteries included" async library, but there are alternatives like trio and curio. MicroPython has `uasyncio` that strives to be compatible with asyncio, but is is really much more complex that what we need in Pybricks. So suppose we roll our own `sagas` module (I'm using the names from redux-sagas for easy cross-referencing for now, but of course don't have to match exactly - if you are paying attention, you can see there are direct correlations to the `tasks` module that Laurens has proposed above, just with different names).

In redux-sagas terminology, a saga is basically a generator function and an effect is basically something that operates on a stream of events. Events are called actions in redux-sagas.

Like all async libraries, we have a main run loop:

```python

def run(saga):
    """
    Runs a saga until completion.
    
    Args:
        saga: any iterable, but most likely a generator/coroutine
    """
    for _ in saga:
        # maybe gc.collect()?
        # wait is particularity needed on ev3dev to allow OS time to run
        # the time could also be a variable to ensure a constant loop time
        wait(1) 
```

Then there are two primitives for combining sagas/effects.

```python

async def all(*effects):
    """
    Runs all effects in parallel until all effects have completed. (similar to asyncio.gather)

    Returns: a tuple of the return value of each effect in the same order as the arguments

    Example::

        sagas.run(sagas.all(saga1, saga2))
[... 85 lines cut ...]

Since many sensors have multiple operating modes for reading different kinds of values, we would need a way to specify which value to read. To do this, we could a an optional `iter_func` parameter to the constructor. This would take a function that takes the sensor object as an argument. This way, you could pass (unbound) sensor methods directly in most cases or make your own more complex function if needed.

```python
ultrasonic = UltrasonicSensor(iter_func=UltrasonicSensor.distance)

for distance in ultrasonic:
    ...
    wait(10)
```

### async mode

Laurens has proposed something like `set_async_mode(True)` to set a global async mode. Riffing on that, here are some variations. 

- Instead of having a separate function, the main loop (such as the `sagas.run()` above) should set this mode and clear it when done.
- When in async mode, calling any blocking function will raise an exception. For example, calling `wait()` would raise an exception that says "use sagas.delay() instead".
- It has been proposed that async mode should change the return value of blocking methods and make them non-blocking when active, but I think we need to consider that this would not work well with autocompletion and other static analysis tools and would make the documentation more verbose because you have to explain that there are overloads that depend on a global state rather than the actual arguments provided. 
  - For the case of wait, I would suggest having a separate function as Python does with `time.sleep()` vs  `asyncio.sleep()`.
[... cut ...]
    - Another option could be to make the motor object itself iterable and always return the motor object no matter what `wait` is. This could then be used to chain motor commands in blocking mode in addition to awaiting in non-blocking mode.
  - If we want to be strict about sensors, we could make methods that read a sensor value that would require a mode change raise an exception in async mode but provide an awaitable `ensure_mode()` method to handle the rare cases (more common on older MINDSTORMS) where a sensor needs to be used in multiple modes in a single program.

laurensvalk, 2023-01-20:
After brainstorming about this today, we realized our ideas align quite closely. I have now updated the example above with a few changes based on this:
- Renamed a few functions to match @dlech's suggestions above more closely.
- Used `task.wait` instead of making `wait` dual-purpose.
- Used an explicit run loop (taking a single generator) instead of using a decorator for main.
- No "hidden" switch for async on import. Rather, it uses async mode only within the aforementioned run loop. This is clearer, and it also allows old-fashioned blocking code to work normally if you want to use it.

laurensvalk, 2023-03-09:
I have updated the `multitasking` branch implementation to reflect recent offline brainstorms. The example above is updated to match.

@dlech if you like you could add your Python implementations of `all`, `race`, etc. to [`bricks/_common/modules/_task.py`](https://github.com/pybricks/pybricks-micropython/blob/multitasking/bricks/_common/modules/_task.py)

As shown in the example above, doing so makes them available as `from pybricks.tools import task`.

There is probably some optimization we could do to make the code size smaller, but this should provide a good base line to evaluate the current approach.

laurensvalk, 2023-05-04:
Updated the `multitasking` branch and did some further exploration.

I was able to get the build size down by 500 bytes by using a common awaitable type instead of giving each class (or even each method) its own wait type. This also makes the types easier to follow.

Instead, you pass this common awaitable these type-specific functions:

```c
/**
 * Tests if awaitable operation is complete. Returns MP_OBJ_STOP_ITERATION if
 * done, possibly with return argument, else mp_const_none.
 *
 * On completion, this function is expected to close/stop hardware
 * operations as needed (hold a motor, etc.). This is not the same as cancel
 * below, which always stops the relevant hardware (i.e. always coast).
 */
typedef mp_obj_t (*pb_awaitable_test_completion_t)(mp_obj_t obj);

/**
 * Called on cancel/close. Used to stop hardware operation in unhandled
 * conditions.
 */
typedef void (*pb_awaitable_cancel_t)(mp_obj_t obj);

```
Each method gets a bit smaller too. Instead of having all sorts of completion logic at the end of each method, it just becomes:
```c
return pb_type_tools_await_or_wait(self, common_Motor_test_completion, common_Motor_cancel);
```

laurensvalk, 2023-05-04:
As another note on build size: Including the tasks module as frozen code costs 852 bytes. Including at as an MPY with a regular script costs 623 bytes. As the module grows, there will probably be a point where freezing is cheaper, but we seem to be pulling in quite bit of extra code now to support the frozen module.

laurensvalk, 2023-05-04:
The total diff is now reduced by -1K to about +1.2K total for the multitasking features so far, which does look good.

marc-hg, 2023-07-03:
I tried running the example code from the original post but it says "cannot import name task". Using spike prime

laurensvalk, 2023-07-03:
Thanks @marc-hg. This example in this exploratory development post was outdated. I've updated it just in case somebody else finds it too. Feel free to try it.

We're working on including this brand new feature in the documentation of the upcoming release. Thanks for giving it a try already!

closed 2023-07-03 by laurensvalk

neout, 2023-10-23:
hi @laurensvalk,
can I ask what happened to this project? is it live? was it replaced by something else?
I cannot open 'multitasking' branch, is it gone?
linked build artifacts cannot be downloaded anymore (their status is 'expired'),
btw. it is unclear to me which artifact should I download (assuming they would work)
I'm new to pybricks, I'm using Robot Inventor (Mindstorms) Hub,
I have already wrote some multitasking code / demo for my robot using this guide:
https://github.com/azzieg/mindstorms-inventor
and I must say multitasking is easy with this, I wanted to transfer my code to pybricks and what I see here more or less matches what I have seen there - however I unable to use this 'pybricks Firmware with multitasking' b/c of the issue described above
ofc. for the demo code above I'm getting import error:
https://github.com/azzieg/mindstorms-inventor

laurensvalk, 2023-10-23:
> is it live?

You can already try all of this live at https://beta.pybricks.com/ (making sure to install the firmware from there, too).

> can I ask what happened to this project? 

A lot of work is happening behind the scenes. Stay tuned for more :)