pybricks/support #918: [Bug]

TortugaSlim, 2023-01-15:
```
Error: Failed to execute 'getPrimaryService' on 'BluetoothRemoteGATTServer': GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
    at https://code.pybricks.com/static/js/main.d3f03113.js:2:396895
    at P.<computed> (https://code.pybricks.com/static/js/main.d3f03113.js:2:1476041)
    at https://code.pybricks.com/static/js/main.d3f03113.js:2:1479953
    at m (https://code.pybricks.com/static/js/main.d3f03113.js:2:1480899)
    at h (https://code.pybricks.com/static/js/main.d3f03113.js:2:1480410)
    at u (https://code.pybricks.com/static/js/main.d3f03113.js:2:1480682)
```

dlech, 2023-01-15:
What operating system are you using? 

Can you share logs as described in https://github.com/pybricks/support/discussions/270?

dlech, 2023-02-13:
Possible duplicate of #312. Likely caused by not compatible Bluetooth adapter.

michalkahle, 2023-04-27:
I can report this bug to happen with my internal bluetooth adapter on Windows after some random time. Output from GetBluetoothRadioInfo:
```
InstanceId        : USB\VID_0A12&PID_0001\6&23B66EB5&0&8
MAC               : 1A7DDA7108
DriverDescription : Generic Bluetooth Radio
DriverVersion     : 10.0.19041.2728
ErrorRecovery     : None
ScoType           : Unknown

InstanceId        : USB\VID_8087&PID_0AA7\6&23B66EB5&0&13
MAC               : 6479F003FBD7
DriverDescription : Intel(R) Wireless Bluetooth(R)
DriverVersion     : 20.100.10.9
ErrorRecovery     : None
ScoType           : Unknown
```

dlech, 2023-04-27:
Thanks for the info. It does seem that builtin Intel Wireless Bluetooth is commonly reported as problematic. Am I correct in understanding that is the one that is not working for you and the one using the generic driver is working?

cc: @juliafishler

michalkahle, 2023-04-28:
Actually, I believe (maybe I'm mistaken, I'll have to check) that there is just single hardware bluetooth adapter in the machine. This particular machine had also problems with the original mindstorms/spike-prime apps connecting to the hubs. We have also 9 other machines in the classroom (different hardware) which have no problems with the original Lego apps. (I'm currently testing pybricks in a single hub with a plan to deploy more widely later.) I'll be able to check the details about the BT adapter next Thursday and I'll report here. I might also check under Linux.

JuliaFiMSFT, 2023-04-28:
Hi Michal, I'm a developer for the Bluetooth team at Microsoft and a passionate Pybricks user. I'd love to help if you are having problems with Bluetooth on your PC. From your comment yesterday, it appears that PC has an internal Bluetooth radio from Intel and a CSR Bluetooth dongle. If that is the case, you'll want to remove or disable one of them. Windows doesn't support two simultaneous Bluetooth radios. Ideally, we would get your internal Intel radio working. If you have a specific problem with it, we can debug that. The best way for me to debug your Bluetooth issues is if you first go to Windows Update and apply all available updates. Then, launch Feedback Hub and use that tool to Report a problem. 
[image]
I recommend adding "Send this bug to JuliaFi" in the title or details so that it gets routed directly to me. 
In the category fields, choose "Devices and Drivers" then "Bluetooth - Other issues"
[image]
The most important part is under attachments, click the "Start recording" then repro the problem you are seeing. When you are done with the repro, click Stop recording and attach the log file. That will ensure that I have all of the logs that I need to debug the issue.
[image]

Let me know if you have any questions!
Julia

MaxPlay812, 2024-05-14:
I have same problem with boost hub 
I fix it with
 restart Bluetooth and disconnect all devices