pybricks/support #921: Publish list of known compatible BLE dongles for Windows

laurensvalk, 2023-01-21:
> Thanks for the followup. Would you mind running this (in powershell) and sharing the results so we can get an idea of which adapters/drivers work and which don't?

_Originally posted by @dlech in https://github.com/pybricks/support/issues/899#issuecomment-1399159557_

laurensvalk, 2023-01-21:
@kwsmit -- since you recently got a few new BLE dongles, would you mind sharing which ones you tried?

Did they work on Windows 10 and/or 11?

laurensvalk, 2023-01-21:
I use this one. It works fine on my dual boot Windows 10 / Ubuntu 22.04 machine.

```
Bus 001 Device 010: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
```

[image]

KWSmit, 2023-01-26:
This `Trust Manga - Bluetooth 4.0`  USB-adapter also works fine on Linux (tested on Pop!_OS 22.04 LTS).

KWSmit, 2023-01-26:
I use this one: `TP-Link UB400 - Bluetooth-adapter - USB - Bluetooth 4.0`. Works fine on Windows (tested on Windows 10) and Linux (tested on Pop!_OS 22.04 LTS)

[image]

Output `GetBluetoothRadioInfo` on Windows:
```
nstanceId        : USB\VID_0A12&PID_0001\5&2F01C3DE&0&1
MAC               : 1A7DDA7115
DriverDescription : Generic Bluetooth Radio
DriverVersion     : 10.0.19041.2486
ErrorRecovery     : None
ScoType           : Unknown

InstanceId        : USB\VID_0A12&PID_0001\5&1F949B48&0&1
MAC               : 1A7DDA7110
DriverDescription : Generic Bluetooth Radio
DriverVersion     : 10.0.19041.2486
ErrorRecovery     : None
ScoType           : Unknown

InstanceId        : USB\VID_33FA&PID_0001\5&2F01C3DE&0&1
MAC               : 47F0E29FB97
DriverDescription : Generic Bluetooth Adapter
DriverVersion     : 10.0.19041.2486
ErrorRecovery     : None
ScoType           : Unknown
```
Output `hciconfig -a` on Linux:
```
hci0:	Type: Primary  Bus: USB
	BD Address: 00:1A:7D:DA:71:10  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:50343 acl:2032 sco:0 events:564 errors:0
	TX bytes:6247 acl:26 sco:0 commands:81 errors:0
	Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'PC-KEES'
	Class: 0x7c0104
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Desktop workstation
	HCI Version: 4.0 (0x6)  Revision: 0x22bb
	LMP Version: 4.0 (0x6)  Subversion: 0x22bb
	Manufacturer: Cambridge Silicon Radio (10)
```

DanieleBenedettelli, 2024-04-09:
I use a TP-link UB500 dongle, that usually works well, but sometimes needs a driver update (like you indicate) or refuses to connect.

Running the command in the PowerShell, it says:

InstanceId        : USB\VID_2357&PID_0604\E848B8C82000
MAC               : E848B8C82000
DriverDescription : TP-Link Bluetooth 5.0 USB Adapter
DriverVersion     : 1.6.1015.3020
ErrorRecovery     : None
ScoType           : Unknown

zaxhs, 2024-08-04:
InstanceId        : USB\VID_0BDA&PID_B00A\00E04C000001
MAC               : 283A4D76A5C8
DriverDescription : Realtek Bluetooth 4.2 Adapter
DriverVersion     : 1.9.1038.3001
ErrorRecovery     : None
ScoType           : Unknown

BertLindeman, 2025-02-13:
I had problems with the internal Intel Wireless Bluetooth adapter.
Most of the times loading a program to a technichub. It stopped at 0 percent or 99% and all between.
The hub is always close at hand. Less than a meter from the PC.
Now I have a `TP-Link UB500` and never had to restart a program load since.
Drivers from TP-Link as documented in the box.