pybricks/support #926: [Question] Will the original (Mindstorms) app still work if I install PyBricks? (Should be FAQ)

blais, 2023-01-29:
**Question**
Will the original (Mindstorms) app still work if I install PyBricks?

**Context**

I discovered the project, and while there's a lot of information, this basic question is not answered and I feel it should be right there at the top of the instructions. Basically, if I install PyBricks, will my kid still be able to use the original Lego-provided app to program the hub?  

I suspect the answer is "no", which would leave me having to do a restore to the original firmware + upgrade frequently for him to use the app (he's too young to do Python yet), and I view that as somewhat of a risky proposition.

In any case, that was one of my most important question when I came to the project, and I couldn't find an obvious answer quickly (well I still don't know for sure). It's really a question about the level of commitment if I install this; if the original app still works, I'm a lot more likely to give it a try, because in the worst case we can still use the original app.

I think it would be useful to add this answer in a prominent place early in the docs.

**Screenshots**
N/A

laurensvalk, 2023-01-29:
Thanks for reminding us about this. We've been focusing on polishing the app, but the website certainly could use more work. We're also moving over the installation pages to a place where they can be edited by contributors (See https://github.com/pybricks/support/issues/780).

To answer your question for now:

_Using the official app requires using the official firmware, so you have to restore the firmware every time you want to use the app._

This is technically not an issue or risk, but I can see it being a bit cumbersome. There are perhaps two ways we can improve to make this easier:

- Restore the latest MINDSTORMS firmware right away. This saves you from doing an update with the MINDSTORMS app. The problem is that we need permission to redistribute that firmware file.
- Add a block language to Pybricks, so there's less of a need to switch back :wink:

blais, 2023-01-29:
Thank you!
(I have the advantage of fresh eyes and ignorance, and perhaps the fear of asking the question while having missed the answer elsewhere on the site... I'm glad I asked! Thanks for your clear response.)

laurensvalk, 2023-01-29:
We can leave this open until we’ve improved it :)

Your feedback is always welcome; please do not hesitate to open new issues as you come across them.