pybricks/support #928: I am unable to get the robot to brake when making turns

TheDoctor2017, 2023-01-30:
**Question**
I am trying to make precise turns using turn "(angle, then=Stop.HOLD, wait=True)". Here is my basic test code:

#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor, TouchSensor, ColorSensor, GyroSensor
from pybricks.parameters import Port, Button, Color, ImageFile, SoundFile, Stop
from pybricks.tools import wait
from pybricks.robotics import DriveBase
import sys
import time

# Initialize the EV3 brick.
ev3 = EV3Brick()

lm = Motor(Port.A)
rm = Motor(Port.D)
wheel_diameter=56
axle_track=125
robot = DriveBase(lm, rm, wheel_diameter, axle_track)
robot.settings(100, 100, 100, 100)
robot.turn(90, then=Stop.BRAKE, wait=True)

When I run the above code, I get the error "TypeError: extra keyword arguments given". I tried different ways of entering the command, but the only one that actually works is robot.turn(90), but this doesn't brake, making the turn actually be greater than 90 degrees. Is there a way to use the "turn" command to get a precise 90 degree turn? Thanks!

laurensvalk, 2023-01-30:
It does stop firmly by default.

But the turn is the last thing in your program. And when the program ends, the motors are released.

Try adding a wait(1000) at the end of your program so you can see it.

To answer your question more literally, the `then` and `wait` arguments are not yet supported in EV3. But they default to hold and True, which are the ones you need.

TheDoctor2017, 2023-01-30:
> But the turn is the last thing in your program. And when the program ends, the motors are released.
     The turn is not the last line in my code, I just didn't post any more of the code because I didn't think it was relevant.

> Try adding a wait(1000) at the end of your program so you can see it.
     Adding a wait made no difference. It still coasts.

> To answer your question more literally, the `then` and `wait` arguments are not yet supported in EV3. But they default to hold and True, which are the ones you need.
     If they are not yet supported, why are they documented that way? And is there a way to set Wait to FALSE in the turn command?

Thanks!

laurensvalk, 2023-01-30:
> If they are not yet supported, why are they documented that way? And is there a way to set Wait to FALSE in the turn command?

The EV3 version of Pybricks is still at 2.0, so it doesn't have all the latest features of Pybricks V3.X, which you may have looked at so far. See https://pybricks.com/ev3-micropython for the EV3 documentation with the matching version.

>  I just didn't post any more of the code because I didn't think it was relevant.

This could help us know a bit more about what is going on.

TheDoctor2017, 2023-01-30:
> The EV3 version of Pybricks is still at 2.0, so it doesn't have all the latest features of Pybricks V3.X, which you may have looked at so far. See https://pybricks.com/ev3-micropython for the EV3 documentation with the matching version.

Do you know when the EV3 version will be updated to V3.X?
> 
> > I just didn't post any more of the code because I didn't think it was relevant.
> 
> This could help us know a bit more about what is going on.

Here is my full test code. It's basically just drive a square:

#!/usr/bin/env pybricks-micropython

from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor, TouchSensor, ColorSensor, GyroSensor
from pybricks.parameters import Port, Button, Color, ImageFile, SoundFile, Stop
from pybricks.tools import wait
from pybricks.robotics import DriveBase
import sys
import time

ev3 = EV3Brick()

color_sensor = ColorSensor(Port.S3)
gyro_sensor = GyroSensor(Port.S2)
lm = Motor(Port.A)
rm = Motor(Port.D)
wheel_diameter=56
axle_track=125
robot = DriveBase(lm, rm, wheel_diameter, axle_track)
robot.settings(100, 100, 100, 100)
robot.straight(360)
wait(1000)
robot.turn(90)
wait(1000)
robot.straight(360)
wait(1000)
robot.turn(90)
wait(1000)
robot.straight(360)
wait(1000)
robot.turn(90)
wait(1000)
robot.straight(360)
wait(1000)
robot.turn(90)
wait(1000)

The robot.strait and robot.turn commands both accelerate and decelerate. I would like to accelerate, but brake on stop. Thoughts?

laurensvalk, 2023-01-30:
As an experiment, can you try the following?

```
robot.turn(90)
wait(1000 * 100) 
```
This will just turn, and then wait a really long time. While it waits, can you manually try to turn the wheels?

Is it holding the wheels in place?

> Do you know when the EV3 version will be updated to V3.X?

There aren't currently any plans. If we can find enough EV3 users who want it, maybe we can make it happen :)

TheDoctor2017, 2023-01-30:
> As an experiment, can you try the following?
> 
> ```
> robot.turn(90)
> wait(1000 * 100) 
> ```
> 
> This will just turn, and then wait a really long time. While it waits, can you manually try to turn the wheels?
> 
> Is it holding the wheels in place?

Yes, it is holding the wheels in place. So it is braking, but it must also be decelerating, which is throwing the angle off. As i mentioned earlier I tried changing the settings line, but it does not affect the acceleration/deceleration of the turns, only the straight. I tried "robot.settings(100, 10000, 100, 10000)", and the straight commands braked properly, but the turn command still accelerates/decelerates.

laurensvalk, 2023-01-30:
> So it is braking

So it is working :smile: 

> but it must also be decelerating, which is throwing the angle off

Acceleration/deceleration are precisely there to avoid throwing it off. So I think what we're seeing that the robot is turning *too far*. This happens when the `axle_track` is too large.

Try measuring it again, and adjust it until it makes the turn work right.

See the _Measuring and validating the robot dimensions_ section on the [Drive Base page](https://pybricks.com/ev3-micropython/robotics.html) for details.

TheDoctor2017, 2023-01-30:
So is there any way with 2.X to set Wait to FALSE?

dlech, 2023-01-30:
You can use [drive()](https://pybricks.com/ev3-micropython/robotics.html#pybricks.robotics.DriveBase.drive) for that.

laurensvalk, 2023-07-13:
The question here has been answered, so we can close this issue.

closed 2023-07-13 by laurensvalk