pybricks/support #930: Is it possible top cycle the power on a particular sensor port?

badassloumd, 2023-01-31:
**Question**
I am dealing with the typical gyro drift issues found on the EV3 gryo sensor. Since there is no known "reset the gyro" command, I was wondering if I can just cycle the power on the port that the gyro is on via python. I see that each sensor port has its own hardware bit that controls power. For example 

I_ONA is tied to the CPU port GP8_bit11
I_ONB is tied to the CPU port GP8_bit12
I_ONC is tied to the CPU port GP8_bit9
I_OND is tied to the CPU port GP6_bit4

   If the gyro was attached to input A, could simply write a zero to GP8_bit11 (shutting off gyro power)
wait a few seconds, and then re-apply power and give it a few seconds to re-cal?

**Context**
 I am trying to reset the Gyron sensor when it is misbehaving

**Screenshots**

[image]
[image]

badassloumd, 2023-01-31:
Never mind.. I see it-- they power the Gyro off of pin 4 -- "VCC5VIGEN"  -- but its not hopeless.  That power is shared among all motor ports and all sensor ports. It is switched on/off via P_EN signal which is attached to P17  GP6 - Bit#5 --- 

[image]

It looks like setting P_EN high will turn on the TPS2041 FET. so this may still be possible.

[image]

dlech, 2023-01-31:
No, it is not possible to cycle the power on a port without turning off power to all ports, the USB host port and the analog/digital converter. The main power for all sensors is supplied on pin 3 (VCC5VIGEN). The I_ONA line provides extra power for certain sensors like the NXT Ultrasonic sensor but is not used on most sensors, including the EV3 Gyro sensor.

> Since there is no known "reset the gyro" command,

There isn't an explicit command that I know of, but if you change the mode of the port (e.g. to "uart-other"), the UART connection will time out and the sensor will reset, then you can change the mode back to "auto" and wait for the sensor driver to load again. On older gyro sensors, changing the mode is enough to cause the sensor to reset, but this only works for certain sensors (there is a date stamp on the sensor you can check) so you could try changing to angle mode and back to speed mode and see if that resets the sensor.

badassloumd, 2023-01-31:
Yeah I see it -- VCC5V is basically the same net as VCC5VIGEN. So that is a poor design choice. Sigh.

"if you change the mode of the port (e.g. to "uart-other"), the UART connection will time out and the sensor will reset, then you can change the mode back to "auto" and wait for the sensor driver to load again. "

   Do you have any Pybrics examples of this code?  

   Thanks.

badassloumd, 2023-01-31:
Did anyone ever try and reverse engineer the gyro serial protocol? I think it might be high time to come up with a better aftermarket gyro..

laurensvalk, 2023-01-31:
I think you can get more or less the effect @dlech described by trying (and failing) to initialize an analog sensor or i2c device or uart sensor: https://pybricks.com/ev3-micropython/iodevices.html#uart-device, and then initialize the gyro again.

laurensvalk, 2023-01-31:
There are several other gyros, see e.g. mindsensors.

dlech, 2023-03-21:
Closing since the question has been answered.

closed 2023-03-21 by dlech