pybricks/support #931: When I try to run a code it does not function

S1icer, 2023-02-02:
**Question**
When I try to run code on my ev3 it just cancels out and does not work

**Context**
I was trying to work with out classes

**Screenshots**

https://user-images.githubusercontent.com/92694741/216206686-dfe71d0d-6841-4e6a-9486-767ab3b52e5a.mov

main py code
https://hastebin.com/share/uvubakemoq.python

dlech, 2023-02-02:
If you run the program from VS Code using the ev3dev extension, you can see the exception and understand the problem. There is probably also a log file on the EV3 that contains the exception.

dlech, 2023-02-02:
Here is a copy of the program from the pastebin link:

```python
#!/usr/bin/env pybricks-micropython

"""
Example LEGO® MINDSTORMS® EV3 Robot Educator Driving Base Program
-----------------------------------------------------------------

This program requires LEGO® EV3 MicroPython v2.0.
Download: https://education.lego.com/en-us/support/mindstorms-ev3/python-for-ev3

Building instructions can be found at:
https://education.lego.com/en-us/support/mindstorms-ev3/building-instructions#robot
"""

import Robot
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import Motor
from pybricks.parameters import Port
from pybricks.robotics import DriveBase
robot = Robot()
robot.upper_medium.reset_angle()

#first variable = distance
#second variable = power (scale is -100n to 100)
robot.move_cms(10,50)

#first variable = roation
#second variable = power 
robot.spin_left(90,50)

#first variable = speed in degrees per second
#second variable = target angle 
robot.upper_medium.run_angle(100,200)

#square up
robot.front_find_and_square()

#find a color, ignore variable 1,2   
#variable 3 = Color.<put in color you want> 
#TURN MOTOR ON BEFORE USING BLOCK
#line 24 starts both large motors
robot.drive.drive(100,0)
robot.two_color_wait(robot.front_left,robot.front_right,Color.Red)
robot.drive.stop()
```

dlech, 2023-02-02:
> ```python
> import Robot
> ...
> robot = Robot()
```

It looks like you are trying to use the class name for the module name or vice versa. If there is `class Robot:` in a file named `robot.py` then you should use `from robot import Robot`.

S1icer, 2023-02-02:
i have a class put in the thing you suggest and use I do have a robot.py file and its still just stalling

dlech, 2023-02-02:
Like I said, you can look at the `main.py.err.log` file or run the program using the ev3dev extension in VS code to see what the error is.