pybricks/support #932: [Bug] Acceleration and angular velocity garbage data

laurensvalk, 2023-02-02:
**Describe the bug**
Something seems to have gone wrong with some of the changes in https://github.com/pybricks/pybricks-micropython/pull/140

**To reproduce**

My hub is stationary, so these values are wrong:

```
>>> hub.imu.acceleration() / 9810
Matrix([
    [  -0.005],
    [  -0.977],
    [  -0.006],
])
>>> hub.imu.angular_velocity()
Matrix([
    [ -35.805],
    [-680.960],
    [ -44.765],
])
```
**Expected behavior**
Acceleration magnitude about 9810, gyro about 0 when stationary.

laurensvalk, 2023-02-02:
This was introduced in https://github.com/pybricks/pybricks-micropython/commit/5942831fb45b906658cb4978dbac1c70d3daeda8, reverting works around it.

dlech, 2023-02-02:
Oh dear. I was working on it so long, by the end I was probably just seeing that the numbers were changing and not looking at the actual values. :disappointed:

dlech, 2023-02-02:
Apparently, this is working correctly on the Technic hub but not on the SPIKE Prime hub.

dlech, 2023-02-02:
It turns out that MicroPython has a quite old STM32F4 HAL (with bugs that have since been fixed upstream) but a newer STM32L4 HAL which is why it worked on Technic hub but not SPIKE.

closed 2023-02-02 by dlech