pybricks/support #934: The editor does not type properly on a Tablet or Mp[Bug]

davek61, 2023-02-03:
**Describe the bug**
What is the problem?

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

davek61, 2023-02-03:
When using the Python editor on a Tablet or Mobile Phone the letters do not appear properly but rather appear to be overlaid on top of themselves.  It can usually be rectified but often additional characters are left on the line.  This is only an issue on a Tablet or Phone.
Because of the nature of this bug it is not possible to provide a screen shot example.

dlech, 2023-02-03:
When you say "Python editor", I assume you mean https://code.pybricks.com?

I've noticed this issue too on my Android 12 phone. I think this has to do with the automatic completion of the keyboard in Android. If I switch to a different keyboard like [this one](https://play.google.com/store/apps/details?id=com.gazlaws.codeboard) the problem goes away.

dlech, 2023-02-03:
[image]

Here is a screenshot showing a missing "r" in "run".

davek61, 2023-02-04:
Hello David,

Thank you very very much for your prompt response and really useful support
information.
I tried that keyboard and it works fine on my phone too now.

This is really helpful and will mean that I can use cheap tablets for
students using Pybricks.

I donated some money to Pybricks yesterday and it was definitely money well
spent! :-)

Kind regards,

David Kempton

dlech, 2023-02-13:
I think the actionable item here is to document the recommendation of https://play.google.com/store/apps/details?id=com.gazlaws.codeboard for Android when an external keyboard is not available.

Info could go in the same place as #921.

closed 2023-02-15 by dlech