pybricks/support #936: Support for bytes encode and decode methods and for sets

ste7anste7an, 2023-02-04:
I want to convert some MicroPython libraries to pybricks running on primehub. Because the current firmware lacks the (cpython) methods encode and decode for strings and bytes, I run into problems. It seems that sets are not supported in the primehub firmware, resulting in compatibitlity issues.

I would like the firmware to be compiled with these MICRO_PY flags set to (1):
```
#define MICROPY_PY_BUILTINS_SET                 (1)
#define MICROPY_CPYTHON_COMPAT                  (1)
```

I build a test firmware with these flags enabled, and the issues I mentioned here got solved. Here is the output of the added methods:
```
>>> set([1,2,3])
{1, 2, 3}
>>> dir(bytes)
['__class__', '__name__', 'count', 'endswith', 'find', 'format', 'index', 'isalpha', 'isdigit', 'islower', 'isspace', 'isupper', 'join', 'lower', 'lstrip', 'replace', 'rfind', 'rindex', 'rsplit', 'rstrip', 'split', 'startswith', 'strip', 'upper', '__bases__', '__dict__', 'decode', 'encode']
>>> s='aaa'
>>> s.encode('utf-8')
b'aaa'
>>> b'1234'.decode('utf-8')
'1234'
```

dlech, 2023-02-04:
Out of curiosity, what do you need these particular methods for?

Related issue that also proposes enabling `set`: #874 

A workaround for `encode` and `decode` is to use `bytes` and `str` instead:

```
>>> bytes("test", "utf-8")
b'test'
>>> str(b"test", "utf-8")
'test'
```

So we could probably get away with not enabling `MICROPY_CPYTHON_COMPAT`.

ste7anste7an, 2023-02-05:
I want to connect an external ESP32-based board (work together with Anton's Mindstorms) to the primehub through one of the sensor/motor ports. That works fine with the SPIKE2 or Inventor Python firmware, where we can enable a UART per port and use that for communication. Now, I thought, I give PyBricks firmware a try. Unfortunately, no native UART is available on the ports, so I started by emulating a PUP device on the ESP32. I can emulate a device with multiple modes. That works fine, I can read and write up to 16 bytes to my emulated PUP device using the `PUPDevice.read` and `.write` methods. We developed a generic communication library as a layer over a serial connection (See [SerialTalk](https://github.com/antonvh/SerialTalk)) and I was porting this library to use the LUMP based communication channel ( see [here](https://github.com/antonvh/SerialTalk/blob/master/serialtalk_lpf2/serialtalk_lpf2.py) for the PyBricks stuff). However, we use a lot of encode/decode methods. I got away with the workaround as you suggested.

 Another nut I have to crack is the way we power the external board. In the normal spike firmware, we use the `hub.port.A.pwm(100)` to output 8V on the motor pin M+ from which we derive 5V (and we use the 3.3V port output to power the logic on the board). I can not get the motor driver enabled in PyBricks without connecting a real motor. Also, I can not switch between Motor and PUPDevice, as that will result in a Device busy error. Is there an alternative way to get power output on a port configured as PUPDevice?

Why are you reluctant to enable `MICRO_CPYTHON_COMPAT`? Limitation of the firmware footprint maybe? Well, there is plenty of space left on the flash storage.

dlech, 2023-02-05:
> Is there an alternative way to get power output on a port configured as PUPDevice?

There are flags that can be set to request constant power on either motor pin. Related code is here:

https://github.com/pybricks/pybricks-micropython/blob/a0676ef2d4bc3e3e94930175c90843e3f6011485/lib/pbio/src/uartdev.c#L892

The flags are encoded as extra data after the mode name as described at: https://github.com/pybricks/technical-info/blob/master/uart-protocol.md#info_name

> Why are you reluctant to enable `MICRO_CPYTHON_COMPAT`? Limitation of the firmware footprint maybe?

Only one person has ever requested it and there is a workaround to accomplish the same thing without enabling it. So there doesn't seem to be a compelling reason to enable it at this time. While there is plenty of flash on the SPIKE hubs, we have to consider the other hubs as well.

ste7anste7an, 2023-02-05:
I saw the constant `PBIO_IODEV_CAPABILITY_FLAG_NEEDS_SUPPLY_PIN1` in the code but could not figure out how to deploy it. Now reading the Lego powered up protocol document again, I see that I totally overlooked the info you mention that goes after the module NAME. I will give that a try!

ste7anste7an, 2023-02-05:
It works! If I carefully emulate a motor-like sensor (not sure what criteria have to be met before the PyBricks hub believes it is a motor) using the same modes as a small mindstorms inventor motor and the same id=75, I can use the name field of the first mode to 'switch' the M+ pin voltage on. The name needs to start with 'POWER'. I use this name field now: `name="POWER\x00\80\x00\x00\x00\x00\x00"` (set bit 7 meaning `requires constant power on pin 2`) and then I measure 7.6 V on the M+ pin of the motor signal. I am still able to use the `PUPDevice.read` and `write` methods to send and receive 16 bytes of data to the external board.

ste7anste7an, 2023-02-12:
It seems that the requirement to emulate a motor with all its modes is not necessary to get power to the M+ motor pin. Only the name should be equal to `POWER` extended with a zero byte and the flags `b'\x00\x80\x00\x00\x00\x00\x00'`. Now we can comfortably power out external devices such as NeoPixels and servo motors from the Hub's battery.

laurensvalk, 2023-02-12:
Yes, see https://github.com/pybricks/pybricks-micropython/blob/280437ab86fd5d060873dca539653253c77aad2d/lib/pbio/src/uartdev.c#L412-L465

It's not even required to call the mode `POWER`. Just the extra data after the name is enough. You can do this for any mode that needs power. In case of Pybricks, we just turn on the power if at least one mode needs it.

If you need power but reversed across the power pins, that is possible too.

And to be sure, make sure your device can handle both ways... Just in case somebody attaches your sensor to the "dumb Technic battery box" and applies power the wrong way.

dlech, 2023-04-28:
I'm going to go ahead and close this as not planned since a suitable workaround was found. We can always reconsider if we get more similar requests in the future.

closed 2023-04-28 by dlech