pybricks/support #938: [Bug] GATT operation failed when screencasting with pybricks

mrinalraghupathi, 2023-02-04:
**Describe the bug**
I am running Mac OS 13.2 Ventura and Chrome 109.0.5414.119 and this is a 2017 Macbook pro with an intel i5.

I am also creating a screencast at the same time of the pybricks app and running the code below. When I run it I get the error 
```
GATT operation failed for unknown reason.
```

I only get this error when the other screencasting software (OBS) is recording. Not sure why?

**To reproduce**
Steps to reproduce the behavior:

Open OBS and start a screen recording of the pybricks application.

Run the following code in pybricks

```
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = InventorHub()

hub.light.on(color=Color.RED)

wait(1000)
```

**Expected behavior**
Run without error (which is what happens when OBS is not running)

laurensvalk, 2023-02-04:
Thanks for reporting.

Does your computer have any other Bluetooth devices connected, like wireless speakers?

mrinalraghupathi, 2023-02-04:
My MX ergo mouse is connected to my laptop and my apple mouse.

dlech, 2023-02-04:
If you turn those off does it work?

To further troubleshoot we will need OS logs such as the Bluetooth packet capture described at https://bleak.readthedocs.io/en/latest/troubleshooting.html#macos

mrinalraghupathi, 2023-02-04:
Indeed turning off the MX ergo mouse and using just the apple mouse fixes the issue it seems. Tried the above run 10 times without error. So that seems to be my solve for now. Awesome. I will capture those logs as well and post so you have that to investigate.