pybricks/support #941: [Feature] Update NXT/EV3 motor parameters

laurensvalk, 2023-02-06:
We updated the motor control algorithms for Pybricks V3, but we did not yet update the control parameters for NXT/EV3 since we are not using them actively.

As a result, they may not work as expected at this time.

We'd need to try this out and update or tune the model parameters as needed.

_Originally posted by @laurensvalk in https://github.com/pybricks/support/discussions/935#discussioncomment-4880137_

winklerp, 2023-02-16:
+ 1

laurensvalk, 2023-02-21:
If #955  is fixed, it looks like comparatively little has to be changed for the parameters.

The main change to look into would be:
- Whether to use the estimated or measured speed (EV3 can get speed fairly accurately)
- Update the KD value, which currently still assumes using measured speed, but it is implemented with estimated speed; this value was never updated.

closed 2023-02-23 by laurensvalk