pybricks/support #968: [Feature] consolidate virtual/physical test scripts

laurensvalk, 2023-02-28:
**Is your feature request related to a problem? Please describe.**
We have `.py` tests in various places at the moment. It could be useful to consolidate them so we don't end up maintaining nearly duplicate tests.

I think we currently have the following:
1. tests/ev3dev: To be run on ev3dev-host
2. tests/pup: Scripts to tests motors, sensors, hubs, API, on a (undocumented) physical SPIKE/Technic build
3. tests/virtualhub: Tests to run on `bricks/virtualhub`
4. tests/motors: Motor data logging tests that can run on any of the above test platforms.

**Describe the solution you'd like**
With some effort it seems like we could at least consolidate 2.--4.

laurensvalk, 2023-02-28:
> on a (undocumented) physical SPIKE/Technic build

Two useful things we can do here:
- Document it with building instructions.
- Adapt the visual in the virtual hub to include the same build.

This should enable us to run the same scripts virtually and on the real hub.

dlech, 2023-03-16:
The most important distinction for me are where/how tests are meant to be run. There currently isn't a clear distinction (my original intention for the `tests` folder was to only contain automated tests for CI/coverage but I think it is OK to use it for everything as long as we make it clear).

Also, recently took me a while to find the pystone test because I never would guess that it would be in the `pup` folder. I think it would be better if the `benchmark` folder was moved up a level.

So maybe something like:

```
tests
+ benchmark
  - pystone.py
  - loop_time.py (future)
+ ci
  + ev3dev
  + virtualhub
+ drive-base
  - some_maneuver.py
  - ...
+ motor-profiling
+ test-stand
  + _common
    + motors
  + ev3 (future)
  + nxt (future)
  + powered-up
    + basic
    + hub
    + sensors
    - README.md (building instructions)

```