pybricks/support #971: [Bug] [pybricksdev] connecting to hub more than once and running program produces error

Novakasa, 2023-03-03:
**Describe the bug**
When trying to connect to a `PybricksHub` more than once (and running a program each time), I get the exception:
```
RuntimeError: disconnected during operation
```

**To reproduce**
PC-side:
```python

import asyncio
from pybricksdev.ble import find_device
from pybricksdev.connections.pybricks import PybricksHub

async def main():

    dev = await find_device()
    print(dev)

    hub = PybricksHub()
    for _ in range(3):
        await hub.connect(dev)
        try:
            print("starting program...")
            await hub.run("brickrail-gui/ble-server/hub_programs/test_noop.py")
            print("program stopped!")
        finally:
            await hub.disconnect()

asyncio.run(main())
```

Hub program:
```python
import usys
from pybricks.tools import wait

print(usys.version)
print("[hub] program start")
wait(30)
print("[hub] program end")
```

Output:
```
1B:BC:1B:A5:81:86: city-green
starting program...
100%|█████████████████████████████████████████████████████████████████████████████████████████████| 213/213 [00:00<00:00, 926B/s]
3.4.0; Pybricks MicroPython v3.2.3 on 2023-02-17
[hub] program start
[hub] program end
program stopped!
starting program...
100%|█████████████████████████████████████████████████████████████████████████████████████████████| 213/213 [00:00<00:00, 887B/s]
Exception has occurred: RuntimeError
disconnected during operation
  File "E:\repos\brickrail\brickrail-gui\ble-server\test_connection_pybricksdev.py", line 16, in main
    await hub.run("brickrail-gui/ble-server/hub_programs/test_noop.py")
  File "E:\repos\brickrail\brickrail-gui\ble-server\test_connection_pybricksdev.py", line 21, in <module>
    asyncio.run(main())
RuntimeError: disconnected during operation
```
The traceback from vscode doesn't track it to the source, here's a traceback from my project that points to the actual point of origin:
```
Task exception was never retrieved
future: <Task finished name='Task-49' coro=<BLEHub.run.<locals>.run_coroutine() done, defined at E:\repos\brickrail\brickrail-gui\ble-server\ble_hub.py:244> exception=RuntimeError('disconnected during operation')>
Traceback (most recent call last):
  File "E:\repos\brickrail\brickrail-gui\ble-server\ble_hub.py", line 247, in run_coroutine
    await self.hub.run(program, print_output=False, wait=True)
  File "E:\repos\brickrail\brickrail-gui\ble-server\.env\lib\site-packages\pybricksdev\connections\pybricks.py", line 374, in run
    await self._wait_for_user_program_stop()
  File "E:\repos\brickrail\brickrail-gui\ble-server\.env\lib\site-packages\pybricksdev\connections\pybricks.py", line 454, in _wait_for_user_program_stop
    await asyncio.wait_for(
  File "E:\repos\brickrail\brickrail-gui\ble-server\.env\lib\asyncio\tasks.py", line 445, in wait_for
    return fut.result()
  File "E:\repos\brickrail\brickrail-gui\ble-server\.env\lib\site-packages\pybricksdev\connections\pybricks.py", line 286, in race_disconnect
    raise RuntimeError("disconnected during operation")
```

I am using the most recent version of pybricksdev from pip, `1.0.0a37`, and the most recent version of the firmware installed from beta pybricks code, `3.4.0; Pybricks MicroPython v3.2.3 on 2023-02-17`

After the exception, the problem continues running without problems, and in my project I don't handle the exception, but the hub actually never disconnects ( the light stays on constantly)

When moving the line
```python
hub = PybricksHub()
```
into the loop, the problem goes away. So I'm guessing the issue must be with some bad state inside `PybricksHub`

Sometimes, a bit of the output of the program is actually printed the second time the program is run, but I can't reproduce that consistently.

**Expected behavior**
I would expect multiple connections in serial to work without any issues.

Novakasa, 2023-03-03:
I reproduced this also with a version of the firmware very close to stable (frozen module added), and pybricksdev `1.0.0a33`

dlech, 2023-03-03:
The problem is that `PybricksHub.disconnect_observable` is currently implemented using an `AsyncSubject` which is basically a oneshot that can't be reset. So it makes `PybricksHub` only good for one connection. We should probably replace it with  a `connection_state_observable` that uses a `BehaviorSubject` to provide the current connection state (disconnected, connecting, connected, disconnecting).

closed 2023-03-03 by dlech

Novakasa, 2023-03-03:
Thanks @dlech for the quick fix! Running the program twice works now, however the disconnect after that still seems to have a problem that seems quite related. Even without running a problem, I am getting stuck after trying to disconnect the second time.

With the cityhub and same firmware as above:
```python

import asyncio
from pybricksdev.ble import find_device
from pybricksdev.connections.pybricks import PybricksHub

async def main():

    dev = await find_device()
    print(dev)

    hub = PybricksHub()
    for _ in range(2):
        print("connecting...")
        await hub.connect(dev)
        print("connected!")
        await asyncio.sleep(1)
        print("disconnecting...")
        await hub.disconnect()
        print("disconnected!")

    print("OK!")

asyncio.run(main())
```

On the second loop, when `disconnect()` is run, the program is stuck for ~2 minutes until finally I get a `TimeoutError`. When I interrupt the program before the timeout it seems that the hub does disconnect quite quickly (compared to usually when my pybricksdev program crashes without disconnecting the hub), so it seems "most" of the disconnect is indeed getting "to the hub".

I have actually seen similar behavior in my project when just connecting once and never running a program, then disconnecting, but never could reproduce it just using pybricksdev.

dlech, 2023-03-03:
Disconnecting Bluetooth devices on Windows seems problematic in general, e.g. https://github.com/hbldh/bleak/issues/1223. Not sure what we can do about other than log everything and try to understand what is going on behind the scenes.

I would add a much longer delay than one second and also add another delay before reconnecting and see if it makes a difference too.

Novakasa, 2023-03-03:
Moving `dev = await find_device()` into the loop doesn't help, I thought that might have been my issue.

Novakasa, 2023-03-03:
I've tested up to a delay of 60 seconds before and after disconnecting, without any success. I also downgraded pybricksdev to 1.0.0a33 because I was surprised I didn't see this problem like this before but it still happens the same. It seems disconnecting always works the first time after I restart the python script though, so it still seems there must be some runtime state somewhere that gets bad. I'll see if I can get some logs over the weekend (will follow your guide about that in discussions).

dlech, 2023-03-03:
Running the example from https://github.com/pybricks/support/issues/971#issuecomment-1454205101 on two different Windows computers here, I'm getting:

```
(.venv) PS C:\Users\david\work\pybricks-issue-971> python .\test.py
1E:FE:CF:EF:58:2B: prime_hub
connecting...
connected!
disconnecting...
disconnected!
connecting...
Traceback (most recent call last):
  File "C:\Users\david\work\pybricks-issue-971\test.py", line 22, in <module>
    asyncio.run(main())
  File "C:\Users\david\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\david\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\david\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 650, in run_until_complete        
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\david\work\pybricks-issue-971\test.py", line 13, in main
    await hub.connect(dev)
  File "C:\Users\david\work\pybricks-issue-971\.venv\Lib\site-packages\pybricksdev\connections\pybricks.py", line 208, in connect  
    await self.client.connect()
  File "C:\Users\david\work\pybricks-issue-971\.venv\Lib\site-packages\bleak\__init__.py", line 471, in connect
    return await self._backend.connect(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\david\work\pybricks-issue-971\.venv\Lib\site-packages\bleak\backends\winrt\client.py", line 413, in connect       
    self.services = get_services_task.result()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\david\work\pybricks-issue-971\.venv\Lib\site-packages\bleak\backends\winrt\client.py", line 643, in get_services  
    services: Sequence[GattDeviceService] = _ensure_success(
                                            ^^^^^^^^^^^^^^^^
  File "C:\Users\david\work\pybricks-issue-971\.venv\Lib\site-packages\bleak\backends\winrt\client.py", line 120, in _ensure_success
    raise BleakError(f"{fail_msg}: Unreachable")
bleak.exc.BleakError: Could not get GATT services: Unreachable
```

Which points to a Windows Bluetooth stack issue (the "Unreachable" error has been reported in Bleak occasionally as well).

dlech, 2023-03-03:
> bleak.exc.BleakError: Could not get GATT services: Unreachable

Moving the scan inside of the loop fixed the issue for me. I forget that the address of the hub changes after each disconnect, so doing this is essential on any OS.

Novakasa, 2023-03-04:
I am familiar with the Unreachable Error, but for me it only shows up if I interfere with the state somehow, e.g. I start the script while the hub is still connected from a previous crashed script. Or just now I tried to start the script too early after booting the system.

Moving `find_device` into the loop doesn't fix it for me, so this might be something on my end.

Novakasa, 2023-03-05:
For completeness here are the wireshark logs and full output of running this script:
```python
import asyncio
from pybricksdev.ble import find_device
from pybricksdev.connections.pybricks import PybricksHub

async def main():

    for _ in range(2):
        dev = await find_device()
        print(dev)
        print("connecting...")
        hub = PybricksHub()
        await hub.connect(dev)
        print("connected!")
        await asyncio.sleep(5)
        print("disconnecting...")
        await hub.disconnect()
        print("disconnected!")
        await asyncio.sleep(5)

    print("OK!")

asyncio.run(main())
```
[pybricks_disconnect_ble_log.pcapng.gz](https://github.com/pybricks/support/files/10891410/pybricks_disconnect_ble_log.pcapng.gz)
[pybricks_disconnect_output.txt](https://github.com/pybricks/support/files/10891411/pybricks_disconnect_output.txt)

For me (complete noob about ble details) it seems interesting (naively) that after the timeout in the end there are some "Disconnect complete" events, so for some reason the disconnect seems to work in the "cleanup" phase?

Novakasa, 2023-03-05:
So I did a bunch more testing with my ~2014 Win10 Laptop and things look a bit different.
While on my desktop, disconnecting always works in the first loop and always fails in the second loop, on my laptop there seems to be a "chance" of the disconnect working for the whole runtime. So If I start the script, either it already fails on the first disconnect, or it works for all the disconnects in my loop (tested with 5 iterations).

I have a suspicion that printing stuff from the hub program and sending data to the hub via `hub.write` increases the chances of the disconnect working, but it also just might be circumstantial.
I have a test running on the laptop with lots of bidirectional communication and that seems to always work without problems.

Another caveat is that on my Desktop I have also seen in a very different application that the disconnect already fails on the first "iteration" if I never run a program on the hub (thus never doing much ble communication as well). I have yet to reproduce this behavior just using pybricksdev.

At this point this feels like a pretty different issue to what I described in the initial post, so I should probably start another issue about this, or maybe we should move the past few posts out to a new issue.

dlech, 2023-03-06:
The problem is possibly with the Bleak library (e.g. maybe something like https://github.com/hbldh/bleak/issues/1249) or just one of many problematic Bluetooth radios/drivers on Windows.

Novakasa, 2023-03-09:
Is there a way to "force" the disconnect on the PC-side? E.g. when my python program crashes for some reason, the hub disconnects eventually. In the same spirit, can I somehow make bleak "unresponsive" to the device as a workaround to this issue?

dlech, 2023-03-09:
I would try installing the `develop` branch of Bleak to make sure it isn't fixed already upstream. If it is still not working there, then enable debug logging so we can see what is going on.

Novakasa, 2023-03-22:
I tested on my desktop with bleak 0.20.0 and all my issues seem to have been fixed 👍 

Since pybricksdev requires <0.20.0, doing `pip install pybricksdev -U` downgrades bleak to 0.19.5, which exhibits all issues I described here.

dlech, 2023-03-22:
Thanks for the feedback. I just updated the dependency and released pybricksdev v1.0.0-alpha.40.

Novakasa, 2023-03-22:
Great, thanks for the quick change!

One bleak/bluetooth-related thing I am still noticing, not sure if this is actually an issue:
If two hubs are connected, the program download is significantly slower for the first device to be connected. Even with both hubs connected, downloading a program to the second connected hub is still fast. Disconnecting the second device makes the program download fast again for the first device.

This is a log from my program, showcasing the difference in download speeds. The fast downloads report as ~1.15kB/s, the slow ones as ~350B/s.
```cwd: E:\repos\brickrail\brickrail-gui
serving now
listening for commands now
[BLEServer] waiting for messages to receive...
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":null,"func":"add_hub","args":["city-blue","smart_train"],"return_id":null}'
[BLEServer] got message: b'{"hub":null,"func":"add_hub","args":["blue","layout_controller"],"return_id":null}'
[BLEServer] got message: b'{"hub":"blue","func":"connect","args":[],"return_id":null}'
[BLEServer] sending message: {"key": "connected", "hub": "blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"city-blue","func":"connect","args":[],"return_id":null}'
[BLEServer] sending message: {"key": "connected", "hub": "city-blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"blue","func":"run","args":[],"return_id":null}'
100%|███████████████████████████████████████████████████████████████████████████████| 1.18k/1.18k [00:03<00:00, 346B/s]
[BLEServer] sending message: {"key": "program_started", "hub": "blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"city-blue","func":"run","args":[],"return_id":null}'
100%|█████████████████████████████████████████████████████████████████████████████| 2.92k/2.92k [00:02<00:00, 1.18kB/s]
[BLEServer] sending message: {"key": "program_started", "hub": "city-blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"city-blue","func":"stop_program","args":[],"return_id":null}'
[BLEServer] sending message: {"key": "program_stopped", "hub": "city-blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"city-blue","func":"run","args":[],"return_id":null}'
100%|█████████████████████████████████████████████████████████████████████████████| 2.92k/2.92k [00:02<00:00, 1.13kB/s]
[BLEServer] sending message: {"key": "program_started", "hub": "city-blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"city-blue","func":"stop_program","args":[],"return_id":null}'
[BLEServer] sending message: {"key": "program_stopped", "hub": "city-blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"city-blue","func":"disconnect","args":[],"return_id":null}'
[BLEServer] sending message: {"key": "disconnected", "hub": "city-blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"blue","func":"stop_program","args":[],"return_id":null}'
[BLEServer] sending message: {"key": "program_stopped", "hub": "blue", "data": null}
[BLEServer] waiting for messages to send...
[BLEServer] got message: b'{"hub":"blue","func":"run","args":[],"return_id":null}'
100%|█████████████████████████████████████████████████████████████████████████████| 1.18k/1.18k [00:01<00:00, 1.15kB/s]
[BLEServer] sending message: {"key": "program_started", "hub": "blue", "data": null}
[BLEServer] waiting for messages to send...
```