pybricks/support #975: [Bug] code.pybricks.com does not overwrite exsisting file on import if requested to overwrite

BertLindeman, 2023-03-09:
**Describe the bug**
Import of file already exsisting in beta.pybricks.com neatly shows dialog.
But a press on the red replace button does not cause the imported file to be really loaded.

**To reproduce**

In beta.pybricks.com add a new program:
``` python
print("original program")
```

Save the program in `Z:\home\bert\py\pybricks\issue`
as `try_overwrite.py`

Update the program in beta.pybricks.com like this:
``` python
print("original program")
print("after save: updated")
```

Now import the original saved program.

dialog shows: 
[image]

Press the red replace button.
No error shown.

Close  `try_overwrite.py`  in the active tab and
re-open `try_overwrite.py` and see still the updated source, **not** the original saved one (as was saved).

Screenshot:
[image]

**Expected behavior**
See newly imported version of source.

**Environment**
Pybricks v3.3.0b2 (Pybricks Beta v2.2.0-beta.1)
Windows 11 Ver 22H2
Microsoft Edge Version 110.0.1587.63 (Official build) (64-bit)

BertLindeman, 2023-03-09:
Huh??

If I set and unset the "remember" flag on the dialog and **then** press the red replace button, **then** the replace works.

If I do not touch the "remember" flag on the dialog, the replace is not done, each time.
No message seen in developers console.

BertLindeman, 2023-03-09:
If the program is open as programming tab then the replace also does not work
regardless op the remember flag.
(hope this is a bit understandable jargon)

dlech, 2023-06-02:
I think I got this fixed in Pybricks Code v2.2.0-beta.5. At least the case replacing a file that is open in the editor. Let us know if you still have other issues with this.

BertLindeman, 2023-06-02:
Works for me.
If asking to overwrite, file is overwritten like requested, regardless the remember-flag.
First tested if open in the editor.
Also if the file is NOT open in the editor, all goes as expected, too.

I am just not sure that the remember flag is indeed remembered.
But this issue has been solved at Pybricks Code v2.2.0-beta.5.

And not related but good to see copied from the status bar:
<html>
<body>

Connected to: | spikeBeta
-- | --
Hub type: | Prime hub
Firmware: | v3.3.0b5

</body>
</html>

closed 2023-06-02 by BertLindeman