pybricks/support #990: Missing peripherals on virtualhub

strongmindsram, 2023-03-16:
When using the built `virtualhub-micropython` executable to run a python application, the virtualhub does not currently have an implementation of either display, imu or light. I can see in the code that support for light was begun but commented out again.

Is this a priority or something that's easy to implement? Sadly I am no shark when it comes to C. I even tried taking a shot at it, but I had to forfeit.

I would love to be able to run my application on a virtual hub in a CI/CD workflow!

laurensvalk, 2023-03-16:
We mainly use the `virtualhub-micropython` to experiment with motors, including live animation as shown below.

If you're interested to see this developed further with display/lights/sensors for use by Strongminds, feel free to reach out at team@pybricks.com --- we're always happy to see what we can do to help.

https://user-images.githubusercontent.com/12326241/221375380-f7f759c5-763b-45b2-a3ea-6d09629c1556.mp4

laurensvalk, 2023-06-01:
Hi @strongmindsram - we've been working on refactoring the sensor driver code, so supporting sensors in the virtual hub is now a bit more feasible.

What kind of sensors would you like to see supported?

strongmindsram, 2023-06-02:
Hi @laurensvalk,

Sounds awesome! The specific needs I had when I looked into it last time was the UltrasonicSensor and the ColorSensor. Of course the most ideal scenario would be for all peripherals to be mockable/supported by the virtualhub, but I understand that things take time.

Also as I mentioned in the original issue, I discovered that there was no support for the matrix display either. It would be cool if one could use the virtualhub even though the peripherals, such as display or imu, does not work. Currently I will get an error if I try to access these features of the virtual hub.

laurensvalk, 2023-06-02:
And to make it more flexible, perhaps we could let the mocked peripherals take inputs/outputs via a socket, so you can make it interface with your "virtual world" app in your own way...

Would this make it more useful for your application?

> The specific needs I had when I looked into it last time was the UltrasonicSensor and the ColorSensor.

Nice. We're quite a bit closer to this now.

Basically, we're generalizing the sensor interface so it will work the same on all platforms (or hubs). Then each platform only has to implement the low-level driver that provides the values. In the virtual hub case, it could return a constant value in the simplest case, or read the value from a socket as mentioned above.

strongmindsram, 2023-06-02:
The most basic use case I had was to simply let my Python application compile and start using your pybricks-micropython distribution to check that the code would actually run on the real hub. It would be awesome if it was possible to also mock input and simulate the real world for greater testing!

laurensvalk, 2023-06-02:
Sounds good! I've sent some further information to the email listed on your organization profile :)