pybricks/support #994: [Bug] Code font on mobile too large

westurner, 2023-03-17:
**Describe the bug**
The code font is too large on a mobile browser (Chrome,) and mobile browsers don't have configurable font sizes

It could be responsive and/or configurable.

- [ ] BUG: CSS: font-size
- [ ] ENH,UBY: CSS,JS: configurable font size

**To reproduce**
Steps to reproduce the behavior:
1. Open a .py document on a mobile device in portrait mode

**Expected behavior**
What did you expect to happen instead?

Font scaling

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.
[image]

dlech, 2023-04-28:
As a workaround you can tell the browser to use the desktop version of the site.

Our minimum requirements for editing is a device with a 10" screen. We try to make sure you can download and run a program on Android phones but it isn't really practical to try to support editing on such small screens.

closed 2023-04-28 by dlech

westurner, 2023-04-28:
FWICS it would otherwise work in landscape mode with an external keyboard.

A responsive CSS breakpoint for smaller screens would solve, too; though configurable font-size is probably also a win for accessibility

westurner, 2023-04-28:
(and also W3C Gamepad API Gamepads attached to pybricks on a phone would be neat; though there is an official LEGO controller that already works FWIU.