pybricks/support #995: [Feature] W3C Gamepad API and controller tester

westurner, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
- drive a LEGO machine with a Gameoad
  - drive a wheeled LEGO vehicle with a gamepad 

**Describe the solution you'd like**
Users can send input from the W3C Gamepad API (and/or directly attached devices) to an event-handling process in order to control a circuit with a gamepad.

**Describe alternatives you've considered**
- https://www.google.com/search?q=%22gamepad+api%22+%22nextjs%22
- https://github.com/topics/gamepad-api?l=typescript&o=asc&s=stars
- https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API
- https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API/Using_the_Gamepad_API
- https://developer.mozilla.org/en-US/docs/Games/Techniques/Controls_Gamepad_API#implementation

**Additional context**
Add any other context or screenshots about the feature request here.

dlech, 2023-03-22:
I think most people would rather connect the gamepad directly to the hub as in #191.

westurner, 2023-04-02:
Where do they pair the controller with the hub? It may or may not make sense to provide a UI in pybricks-code to get the BLE information and remap buttons then transfer that config to the hub?

westurner, 2023-04-02:
- :eyes: More notes here: https://github.com/pybricks/support/issues/262#issuecomment-1493283886

laurensvalk, 2023-04-02:
Whoops, I made a mistake in trying to move this post. Please do keep discussing. We're just trying to keep things easy to find :smile: