pybricks/support #1000: [Question] Syn PybricksHub Class

BY571, 2023-03-21:
**Question**
Hey, I created this class to communicate with the PybricksHub in a synchronous way and keep the asynchronous execution. Communication works well, however, I run into two issues. Sometimes I get disconnects randomly and when trying to disconnect manually I get an error message.  

**Context**
Regarding the random disconnects (that's what I would call them) I noticed that first the byte size changes and the received data in the read function returns the following error message in the printouts: 
[image]
You can see it runs first correctly receiving data of size 73 (bytes) with length 16 but then the error message gets read and the byte size changes to 20 (due to the error message getting sent). The entire error message: `KeyboardInterrupt: Pybricks MicroPython v3.2.0b6 on 2022-12-02; SPIKE Prime Hub with STM32F413VG Type "help()" for more information`.

Interestingly is that I can once this happens press the start/stop button to disconnect and press it again to connect again and the robot runs again and executes the script. However, it's not in line with the printouts I receive. It seems like previous actions are queued up and then directly executed once the hub connects again.  These "disconnects" with the error message can happen several times in one "run" of the main script. But I also had several runs in a row (3-4) without any problem.

For the second problem when finishing the main script, I get the following error message:
[image]

When I run my script within a single async function like you showed in [pc-communication](https://pybricks.com/projects/tutorials/wireless/hub-to-device/pc-communication/ ) everything works well so I guess both problems are related to my class.

This is the class you can see I tried to reuse many of the code mentioned in the pc-communication example:
`import asyncio
from bleak import BleakScanner, BleakClient
from bleak.backends.device import BLEDevice
import struct

class PybricksHub:
    """ Class for connecting to a Pybricks Hub. """
    UART_SERVICE_UUID = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
    UART_RX_CHAR_UUID = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
    UART_TX_CHAR_UUID = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
    HUB_NAME = "Pybricks Hub"

    def __init__(self, out_format_str: str, state_dim: int):
        self.device = None
        self.client = None
        self.rx_char = None
        self.rx_queue = asyncio.LifoQueue(maxsize=5) # Queue
        self.loop = asyncio.get_event_loop()
        
        self.exception_out_data = struct.pack(out_format_str, *([0.0] * state_dim))
        
    def connect(self)-> None:
        """ Connect to the hub. """
        print("Connecting to the hub...")
        self.loop.run_until_complete(self._connect())

    async def _connect(self)-> None:
        """ Connect to the hub. """
        try:
            # Find the device and initialize client.
            self.device = await BleakScanner.find_device_by_filter(self._hub_filter)
            self.client = BleakClient(self.device, disconnected_callback=self._handle_disconnect)

            # Connect and get services
            print("Switch on the hub")
            await self.client.connect()
            await self.client.start_notify(self.UART_TX_CHAR_UUID, self._handle_rx)
            nus = self.client.services.get_service(self.UART_SERVICE_UUID)
            self.rx_char = nus.get_characteristic(self.UART_RX_CHAR_UUID)

            # Tell user to start program on the hub.
            print("Start the program on the hub now with the button.", flush=True)
            await asyncio.sleep(5)

        except Exception as e:
            # Handle exceptions.
            print(e)
            await self.disconnect()

    def send(self, data: bytes)-> None:
        """ Send data to the hub as bytes. """
        self.loop.run_until_complete(self._send(data))

    async def _send(self, data: bytes)-> None:
        try:
            # Send some data to the hub.
            await self.client.write_gatt_char(self.rx_char, data)
        except Exception as e:
            # Handle exceptions.
            print(e)
            await self.disconnect()

    def disconnect(self)-> None:
        for task in asyncio.all_tasks():
            task.cancel()
        # self.loop.run_until_complete(self._disconnect())
        # self.loop.close()

    async def _disconnect(self)-> None:
        try:
            # Disconnect when we are done.
            if self.client:
                await self.client.disconnect()
                self.client = None
                self.device = None
                self.rx_char = None
        except Exception as e:
            # Handle exceptions.
            print(e)
        finally:
            self.disconnected = True

    @staticmethod
    def _hub_filter(device: BLEDevice, ad)-> None:
        return device.name and device.name.lower() == PybricksHub.HUB_NAME.lower()

    def _handle_disconnect(self, _)-> None:
        print("Hub was disconnected.")
        self.disconnect()

    async def _handle_rx(self, _, data: bytes)-> None:
        # add received data to the queue
        await self.rx_queue.put(data)
    
    async def _read_data(self)-> bytes:
        try:
            # get data from the queue
            return await self.rx_queue.get()
        except asyncio.QueueEmpty:
            print("Queue is empty, returning zeros")
            return self.exception_out_data

    
    def read(self)-> bytes:
        """Read data from the hub and return it as a bytearray."""       
        return self.loop.run_until_complete(self._read_data())

    def close(self)-> None:
        # TODO: this is not working correctly -- "never awaited disconnect"
        # self.disconnect()
        # Disconnect from the hub.
        self.disconnect()
        
        # self.loop.run_until_complete(self._disconnect())

        # # Wait for the disconnection to complete.
        # timeout = 5  # seconds
        # start_time = self.loop.time()
        # while not self.disconnected:
        #     self.loop.run_until_complete(asyncio.sleep(0.1))
        #     if self.loop.time() - start_time > timeout:
        #         print("Timed out waiting for disconnection to complete.")
        #         break

        # # Close the event loop.
        # self.loop.close()

`
I included some commented-out parts to see what else I have tried to get a disconnect without error.

dlech, 2023-03-21:
> The entire error message: `KeyboardInterrupt: Pybricks MicroPython v3.2.0b6 on 2022-12-02; SPIKE Prime Hub with STM32F413VG Type "help()" for more information`.

If you are sending binary data over stdio, you will need to disable the keyboard interrupt in the program running on the hub, otherwise `b'\x03'`  will trigger the `KeyboardInterrupt`. 

https://docs.pybricks.com/en/stable/micropython/micropython.html#micropython.kbd_intr

BY571, 2023-03-22:
> If you are sending binary data over stdio, you will need to disable the keyboard interrupt in the program running on the hub, otherwise `b'\x03'` will trigger the `KeyboardInterrupt`. [...]

Thanks a lot! This indeed solved the problem. 

Regarding the first problem, I agree that it's better to run the whole script like `asyncio.run(main())`, however, my project requires me to send messages back and forth with the hub while running another script synchronously. That's why I had to mess around with the example code in [pc-communication](https://pybricks.com/projects/tutorials/wireless/hub-to-device/pc-communication/) and create a class.

BY571, 2023-03-22:
I found the solution to the other problem. 
Updating the disconnect function like:
```    
def disconnect(self)-> None:
        if self.loop.is_running():
            for task in asyncio.all_tasks(loop=self.loop):
                task.cancel()
            self.loop.run_until_complete(self._disconnect())
            self.loop.close()
```
            
fixed the problem with the error message when closing.

closed 2023-03-22 by BY571