pybricks/support #1008: [Question] Communicate with PC with USB.

Developer-Mike, 2023-03-24:
**Question**
Is it possible to communicate between the EV3-Brick and the PC through USB?

**Context**
I'm trying to make a Mindstorms robot that plays a game using the Minimax algorithm. The problem is that the brick is too slow to calculate those moves.

dlech, 2023-03-24:
Pybricks currently only supports Bluetooth communications. USB is only used for charging the battery. Please see #262 for more info.

Developer-Mike, 2023-03-24:
Thank you for your blazingly fast response. So I guess that [this](https://github.com/pybricks/pybricks-api/tree/master/examples/ev3/bluetooth_pc) is an example that would fit my case.

dlech, 2023-03-24:
Yes, that is the simplest way to use Bluetooth with Pybricks on EV3. 

Although I should also mention on EV3, it is possible to have a network connection over USB too, so you can use basic networks sockets if you know how to do that.

Developer-Mike, 2023-03-24:
Can you elaborate further? What do you mean by a network connection over USB?  Do you mean a connection to the network over the PC?

dlech, 2023-03-24:
Out of the box, ev3dev is set up to create a network connection to the host computer over USB. https://www.ev3dev.org/docs/tutorials/connecting-to-the-internet-via-usb/

This is what allows connecting to the EV3 via SSH either for a remote terminal or transferring files with SCP.

But since it is just a basic network connection, you can do TCP/IP, e.g. using the [MicroPython socket module](https://docs.micropython.org/en/latest/library/socket.html).

Developer-Mike, 2023-03-24:
Thank you so much. Because I'm connected to the network of the PC the IP address for connecting from the brick to the computer should be 127.0.0.1. Correct?

Developer-Mike, 2023-03-24:
Worked! Thank you so much 😊

closed 2023-03-24 by Developer-Mike